LazeMSS/OctoPrint-UICustomizer #188: Setting Icon not able to be selected

kelvindavis, 2021-09-01:
After todays update to UI Customizer, I am unable to select the setting icon (wrench).  If I go into safe mode and turn off UI Customizer, I am able to select it as normal.  Was working fine up until todays update.

LazeMSS, 2021-09-01:
Do you have responsive mode enabled or not?

endurance1968, 2021-09-01:
Same issue here. I updated on 2 of three printers - the updated ones are showing the issue the one on previous version (1.6.4) is working as expected.
Where to enable/disable responsive mode?

LazeMSS, 2021-09-01:
In settings :/ Working on a fix - but try making the window wider - this should fix and allow settings tool.

endurance1968, 2021-09-01:
Oh great in settings :)
Wider does not help -even more than 3400 pixels wide, also shrinking to min and chnaging back to wide does not change the behaviour here.

LazeMSS, 2021-09-01:
That's really strange - what browser?

kelvindavis, 2021-09-01:
Wider did not work for me and I have used Safari and chrome

endurance1968, 2021-09-01:
FF - but making it smaller so you get the menu instead of navbar items allows you to access the settings.

LazeMSS, 2021-09-01:
Working fine on chrome for me - will test in FF

LazeMSS, 2021-09-01:
@endurance1968  could you try and go into settings - and UI Customizer settings and click report a bug so I can get the data

kelvindavis, 2021-09-01:
I can do that as well after the print I’m doing finishes

endurance1968, 2021-09-01:
deactivating resp mode works - but you have to refresh browser manually as all buttons are gone otherwise.

## Description
**ENTER DESCRIPTION HERE
Describe your problem?
What is the problem?
Can you recreate it?
Did you try disabling plugins?
Did you remember to update the subject?**
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.6.6**
- Top Temp[toptemp] v0.0.1.4
- Translate Model Plugin[translatemodel] v0.2.0
- Printer Stats[stats] v3.1.2
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- Draggable Files[draggable_files] v1.0.4
- Dashboard[dashboard] v1.18.3
- Octolapse[octolapse] v0.4.1
- OctoKlipper[klipper] v0.3.8.3
- Filament Sensor Simplified[filamentsensorsimplified] v0.2.3
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.26.0
- Bed Visualizer[bedlevelvisualizer] v1.1.0

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: false
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.6.1
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:91.0) Gecko/20100101 Firefox/91.0

LazeMSS, 2021-09-01:
Do you know what theme your running and can you post a screenshot maybe?

endurance1968, 2021-09-01:
Screenshot in a few miinutes

Second printer
**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.6.6**
- Top Temp[toptemp] v0.0.1.4
- Themeify[themeify] v1.2.2
- Translate Model Plugin[translatemodel] v0.2.0
- Tab Order[taborder] v0.5.12
- Printer Stats[stats] v3.1.2
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- LightsOut[lightsout] v0.1.2
- Fix CBD Firmware Plugin[fixcbdfirmware] v0.4.0
- Filament Sensor Reloaded[filamentreload] v1.3.4
- Draggable Files[draggable_files] v1.0.4
- Dashboard[dashboard] v1.18.3
- Octolapse[octolapse] v0.4.1
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.26.0

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.6.1
- Python 3.7.3
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:91.0) Gecko/20100101 Firefox/91.0

LazeMSS, 2021-09-01:
Btw. Themify and Tab Order should not be needed

kelvindavis, 2021-09-01:
I was using slate

endurance1968, 2021-09-01:
[image]
[image]

endurance1968, 2021-09-01:
>Btw. Themify and Tab Order should not be needed
okido - will deinstall

LazeMSS, 2021-09-01:
@endurance1968  can you post a screenshot of the main ui ?
And if possible run this command from the developer console (F12):
$('div.UIHeaderWrap').css('background-color','#ff0000')

kelvindavis, 2021-09-01:
I have joined it but new to discord

LazeMSS, 2021-09-01:
@kelvindavis what name?

kelvindavis, 2021-09-01:
OHKinkBear

endurance1968, 2021-09-01:
I just joined discord.
[image]

LazeMSS, 2021-09-01:
yeah seems like something is wrong with the width of that overlay

LazeMSS, 2021-09-01:
@endurance1968 do you have a user name?

endurance1968, 2021-09-01:
endurance

LazeMSS, 2021-09-01:
The bug has been found - long usernames DOOH

endurance1968, 2021-09-01:
happy to test as soon as you have the update ready

kelvindavis, 2021-09-01:
Let us know when and be happy to test

LazeMSS, 2021-09-01:
Fixed in next release

LazeMSS, 2021-09-01:
Out now - 0.1.6.7

closed 2021-09-01 by LazeMSS

endurance1968, 2021-09-01:
Updated - and seems to work - thx