LazeMSS/OctoPrint-UICustomizer #192: Mobile interface instead of the desktop

ChristopheBis, 2021-09-07:
## Description

With the last update of this pugin, the interface of octoprint show the responsive version (everything full width in one column)
Did you try disabling plugins? Yes and everything is fine, but the issue is back as soon as the plugin is enabled.

<hr>

**Plugins installed**
- Z Probe Offset[z_probe_offset] v0.5
- Upload Anything[uploadanything] v1.0.1
- **UI Customizer[uicustomizer] v0.1.6.8**
- PrintJobHistory[PrintJobHistory] v1.14.0
- Preheat Button[preheat] v0.8.0
- Cura Thumbnails[UltimakerFormatPackage] v1.0.1
- SpoolManager Plugin[SpoolManager] v1.4.3
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- Printoid Plugin[printoid] v1.0.2
- Navbar Temperature Plugin[navbartemp] v0.14
- MultipleUpload Plugin[MultipleUpload] v1.1.0
- Floating Navbar[floatingnavbar] v0.3.7
- Firmware Updater[firmwareupdater] v1.12.0
- FileManager[filemanager] v0.1.6
- Exclude Region[excluderegion] v0.3.0
- Dashboard[dashboard] v1.18.3
- BLTouch Plugin[BLTouch] v0.3.4
- Marlin EEPROM Editor[eeprom_marlin] v3.1.0
- Macro[macro] v0.3.0
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.26.0
- Bed Visualizer[bedlevelvisualizer] v1.1.0
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.6.1
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/93.0.4577.63 Safari/537.36 Edg/93.0.961.38

spenxerx4, 2021-09-07:
Same issue here. Customizer doesn't save settings. Single column. Its broken :(

ejjenkins, 2021-09-07:
Same for me...I had to disable it.

LazeMSS, 2021-09-08:
Working on it now

LazeMSS, 2021-09-08:
Which versions of OctoPi are you all running - @ChristopheBis  you are on 0.16.1?

LazeMSS, 2021-09-08:
Could anyone of you be so kind and open the developer console and reload and see what errors your are getting there? I cant recreate it.

LazeMSS, 2021-09-08:
BUG FOUND: Fix will be released ASAP - it you select a theme it will work until the new version is released.

LazeMSS, 2021-09-08:
Fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.6.9

If not working please reopen/comment.

closed 2021-09-08 by LazeMSS

ChristopheBis, 2021-09-08:
> Which versions of OctoPi are you all running - @ChristopheBis you are on 0.16.1?

Hi, my octopi version is 0.18.0

Switching to another theme is working, but I can't go back to the default theme. It is selected, but the theme is not applied.

LazeMSS, 2021-09-08:
Try updating to the latest version 0.1.6.9

ChristopheBis, 2021-09-08:
We're back in business!! 

0.1.6.9 is working

Thank you!

ejjenkins, 2021-09-08:
Mine is fixed as well.  Thank you!