LazeMSS/OctoPrint-UICustomizer #193: Updated to 0.1.6.8 Chrome nor Edge will not show multiple columns

bwerst, 2021-09-08:
Updated from 0.1.6.7 to 0.1.6.8 and I lost the three column layout.  I've tried multiple options in the plugin, but not sure where to go next. 

[image]

[image]

longwa, 2021-09-08:
Same issue, Chrome on Mac OS. No columns.

longwa, 2021-09-08:
Looking at the changes in this patch release, I'm surprised this wasn't saved for a larger release. These are definitely some non-trivial changes to the layout to make in a patch release. I feel like changes like this should've been targeted at a 0.1.7.x type branch at least with a bit of beta testing to boot.

https://github.com/LazeMSS/OctoPrint-UICustomizer/compare/0.1.6.7..0.1.6.8

LazeMSS, 2021-09-08:
Not a major version - i normally just release when I have two new features/fixes.

LazeMSS, 2021-09-08:
I have spent a lot of time testing this - the only majjor stuff is a rewrite of the plugin checks

LazeMSS, 2021-09-08:
What version of octoprint are you running?

LazeMSS, 2021-09-08:
BUG FOUND: Fix will be released ASAP - it you select a theme it will work until the new version is released.

LazeMSS, 2021-09-08:
Fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.6.9

If not working please reopen/comment.

closed 2021-09-08 by LazeMSS

bwerst, 2021-09-08:
0.1.6.9 solved the problem.  Thank you!  I really like having the multi columns and the compressed title bar with the key stats.

LazeMSS, 2021-09-08:
Great - also look at toptemp if you haven't tried it