LazeMSS/OctoPrint-UICustomizer #195: [Feature Request] a "don't show again" for a plugin compatibility warning.

GhostlyCrowd, 2021-09-08:
It's giving a false positive trigger on "Stateful Sidebar" which has, always has, and continues to work perfectly fine with UI Customizer in my experience.

It would be great not be bugged about its presence  each time I load Outsprints web interface.

LazeMSS, 2021-09-08:
They collide because UI customizer has the same option - that' why I have added the notice. Maybe I should check if you enable that option in UI Customizer :)

GhostlyCrowd, 2021-09-08:
> They collide because UI customizer has the same option - that' why I have added the notice. Maybe I should check if you enable that option in UI Customizer :)

I suppose a check would work, I wasnt aware UI customizer does the same thing I must have missed that option.

LazeMSS, 2021-09-09:
It's a new feature 😄

LazeMSS, 2021-09-20:
Should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.7.0
Please check and repopen/comment it not working as expected.

closed 2021-09-20 by LazeMSS