LazeMSS/OctoPrint-UICustomizer #196: Tiles not arranged

Buster-Grob, 2021-09-09:
## Description
I move the tiles around in the Plugin settings, and it indicates that its moved them, but doesnt, even after saving and restarting. 

Can you recreate it? This is the seond install this has happened on.
Did you try disabling plugins?  I this is the only plugin that changes theme or appearance.

<hr>

**Plugins installed**
- Cost Estimation[costestimation] v3.3.0
- Dashboard[dashboard] v1.18.3
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.26.0
- Filament Manager[filamentmanager] v1.8.1
- FileManager[filemanager] v0.1.6
- Octolapse[octolapse] v0.4.1
- Preheat Button[preheat] v0.8.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- PSU Control[psucontrol] v1.0.6
- PSU Control - Tasmota[psucontrol_tasmota] v1.0.1
- Resource Monitor[resource_monitor] v0.3.4
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- **UI Customizer[uicustomizer] v0.1.6.9**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.6.1
- Python 3.9.5

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/93.0.4577.63 Safari/537.36

LazeMSS, 2021-09-09:
Tiles? Is that items like files, connection etc?
I can't recreate it right now
Maybe provide a screenshot or two?

Buster-Grob, 2021-09-09:
Yes, the connection, status, files, filament manager, etc. It gives me the
default arrangement, but when I change it in the configuration, it says it
worked, but doesn't change the layout at all. I'll send you some
screenshots tomorrow

LazeMSS, 2021-09-09:
Ohh and please try an open the console:
https://appuals.com/open-browser-console/

And then reload the octoprint interface and post the log.

LazeMSS, 2021-09-11:
@Buster-Grob any news?

Buster-Grob, 2021-09-11:
Hey! Sorry for the late reply

So I ended up having multiple other issues with my OctoPrint install,
mainly related to the computer being old and slow.

I put a fresh install on a VM on my server, reinstalled the UI Customizer
plugin, and I'm having zero issues now.

Sorry I didn't get any screen shots or logs too you. If it happens again,
I'll be sure to grab those. Thank you!

closed 2021-09-11 by LazeMSS