LazeMSS/OctoPrint-UICustomizer #197: Navbar Icons don't display correctly on mobile if they are too wide.

Dr-Ari-Gami, 2021-09-13:
With just a few plugins installed that add navbar indicators, on mobile they run off the screen. It would be nice if if they horizontally scrolled. 

[image]

I'd submit a pr, but It's been a few years since I've done web development.

LazeMSS, 2021-09-13:
I had it working in a release but it caused problems with plugins overflowing with popovers etc.

I'm still working on a solution

electronicm, 2021-09-13:
Would it make sense to determine the height of the div via JS and move all content downwards accordingly?

What I could well imagine would be the same mechanism as is used for the tabs in the main section. So that all icons that exceed the display width are in a dropdown.

However, I see the display of the DisplayLayerProgress plugin as a fundamental problem. Due to the display of long text, this is not suitable for small display areas. Should the developer be involved in solving the problem?

LazeMSS, 2021-09-13:
Yes but it would require a resize event handler - I did 90% of the work in CSS and a bit of js - but hiding the overflow caused the popovers etc. for some plugins to be hidden.

I think I will do a complete rewrite where its handle ind a better way to have the icons in a pull down fasion - still testing a lot of ideas.

NilsRo, 2023-01-04:
Perhaps you can do the same and put all LI elements into a dropdown. Would not looks that nice but the UI will stay accessible.
[image]