LazeMSS/OctoPrint-UICustomizer #200: [Feature Request] Option to "pop out" webcam in a new window (instead of browser fullscreen)

bcat, 2021-09-17:
Hello! Thanks for developing this plugin. It's really useful for improving the OctoPrint user experience.

I wanted to suggest a small tweak to the fullscreen webcam functionality. In addition to the current mode that uses the browser's fullscreen API, would it be possible to pop the webcam out into a new window.  Right now, I either have the webcam taking up an entire monitor (a lot of space) or I have the small camera with the other OctoPrint UI around it. But I'd really love it if I could have the "fullscreen webcam view" in a dedicated window/tab that I could move around accordingly. (Right now I just open the MJPEG stream its own tab manually, but then I lose the nice "status bar" that the UI Customizer plugin adds.)

I know this is a niche use case and I understand if it's too specialized to implement, but I figured I would file the request and see what you thought of it. :)

LazeMSS, 2021-09-17:
On the to-do

bcat, 2021-09-19:
Great, thank you! :)