LazeMSS/OctoPrint-UICustomizer #202: Main Tabs order doesn't work

ThomasCr, 2021-09-23:
## Description
I try to reorder the main tabs, but it get ignored. I also tested the force reload the website / restart browser.
Funny is also, that the icon setup is also ignored, as you can see, that there are no icons.
[image]
[image]

<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.7.0**
- Print Scheduler[printscheduler] v0.0.4
- Resource Monitor[resource_monitor] v0.3.5
- Pushover[pushover] v0.3.8
- PSU Control[psucontrol] v1.0.6
- PSU Control - Shelly[psucontrol_shelly] v1.0.2
- PortLister[portlister] v0.1.10
- Camera Settings[camerasettings] v0.2.2
- Marlin GCode Documentation[marlingcodedocumentation] v0.12.0
- Marlin EEPROM Editor[eeprom_marlin] v3.1.0
- Calibration-Companion[calibrationcompanion] v1.0.1
- Action Commands[actioncommands] v0.4

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: false
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.6.1
- Python 3.8.10

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:92.0) Gecko/20100101 Firefox/92.0

LazeMSS, 2021-09-23:
What happens if you are making changes with preview on - does it update then?

ThomasCr, 2021-09-23:
funny, we had a power outage in our village... now it works how it should.... I don't know what was really the reason.

LazeMSS, 2021-09-23:
Closing - reopen if it happens again...

closed 2021-09-23 by LazeMSS