LazeMSS/OctoPrint-UICustomizer #204: Custom control editor created buttons disappear every time

marcodagostino, 2021-10-01:
## Description
Have you noticed about a bug with UI customizer and custom control editor plugins both activated?
Every refresh of gui page custom buttons created by Custom Control Editor plugin disappear.
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.7.0**
- Telegram Notifications[telegram] v1.6.5
- Detailed Progress[detailedprogress] v0.2.7
- Custom Control Editor[customControl] v0.2.2
- Marlin EEPROM Editor[eeprom_marlin] v3.1.0
- Bed Visualizer[bedlevelvisualizer] v1.1.0
- Action Commands[actioncommands] v0.4

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: true
- themeLocal: false

**Software versions**
- OctoPrint 1.6.1
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/94.0.4606.61 Safari/537.36

LazeMSS, 2021-10-01:
No but I will test ASAP

LazeMSS, 2021-10-02:
Works fint on my installation - can you provide a bit step by step how to recreate?

marcodagostino, 2021-10-02:
Yes.. Sorry guy.. Retrying I realized that with standard ui the bug persists.
It's a issue of the other plugin.
We can close thread too

closed 2021-10-03 by LazeMSS