LazeMSS/OctoPrint-UICustomizer #215: Terminal window will not adjust up

roontoon, 2021-11-01:
## Description
A terminal window will not adjust up
https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/214

Can you recreate it? yes
Did you try disabling plugins?
Did you remember to update the subject?**
<hr>

**Plugins installed**
- Z Probe Offset[z_probe_offset] v0.5
- **UI Customizer[uicustomizer] v0.1.7.0**
- Thermal-Runaway[ThermalRunaway] v3.0.2
- Terminal Commands Extended[terminalcommandsextended] v0.1.6
- Printer Stats[stats] v3.1.2
- Resource Monitor[resource_monitor] v0.3.5
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- Print History Plugin[printhistory] v1.3
- PrettyGCode[prettygcode] v1.2.4
- Exclude Region[excluderegion] v0.3.0
- Active Filters Extended[active_filters_extended] v0.1.0
- Google Drive Backup[googledrivebackup] v0.0.3
- Calibration-Companion[calibrationcompanion] v1.0.1
- Bed Visualizer[bedlevelvisualizer] v1.1.0
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: true
- themeLocal: false

**Software versions**
- OctoPrint 1.7.0
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/95.0.4638.54 Safari/537.36

LazeMSS, 2021-11-04:
Can you test in Chrome?

roontoon, 2021-11-05:
Yes it works in Chrome but not Safari. Also came across another difference which you can see in second screenshot.

LazeMSS, 2021-11-06:
Okay fired up the old mac - Safaris behaving kinda odd - I will try and fix it

roontoon, 2021-11-06:
As is chrome on the last picture with the arrow.

> On Nov 6, 2021, at 07:44, LazeMSS ***@***.***> wrote:
> 
> Okay fired up the old mac - Safaris behaving kinda odd - I will try and fix it
> 
>

LazeMSS, 2021-11-13:
Should now be included in the 0.1.8.0 release (https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.8.0) - please reopen if not working as expected.

closed 2021-11-13 by LazeMSS