LazeMSS/OctoPrint-UICustomizer #216: Plugin Compatibility Issue

PRGeno, 2021-11-04:
Octoprint 1.7.2 
Python 3.7.3
UICustomizer 0.1.7.0
Tab Order 0.5.12
Fullscreen 0.0.6
Themeify 1.2.2

I just upgraded Octoprint from 1.6.1 and Python from 2.17.16 to the latest.  I had to abandon a few old plugins that are not working with Python3 and have everything working well.  However Octoprint pops "Plugin compatibility issue" warnings saying UICustomizer should be disabled due to compatibility issues with three of my remaining plugins:

Tab Order
Fullscreen
Themeify

I haven't seen any issues so far beyond the warning messages and all 4 plugins seems to be working fine together.

I really like what I'm able to do with UICustomizer, but I also find those other 3 plugins essential to my Octoprint environment on all of my printers.  I don't want to disable any of the 4 and with the common plugin being UICustomizer I came here to report the issues.

LazeMSS, 2021-11-04:
The warnings are there because I often get bug reports about problems which are caused by plugins that are not compatible.

Tab order does nothing that UI customizer don't already provide. So you can just uninstall it. 

Full screen is also done by UI customizer.

Same story with Themeify

Unless there is something special these 3 plugins do that I'm not aware of then I can be wrong but I'm 99% sure you can disable the 3 plugins and get the same functionality but just with one plugin.

PRGeno, 2021-11-04:
OK, I haven't dug into the recent features you've added and was likely using a relatively old version.  I've started playing with it again and immediately saw how Tab Order is no longer needed.  Now off to explore to possibility of eliminating Fullscreen and Themeify next.

Thanks for your prompt reply and thanks for creating such a useful plugin

LazeMSS, 2021-11-04:
It is true that at the moment I don't offer any "customizing" of colors to the same extent as Themeify  - but you can do alot by using "advanced" section. Also note that Themeify and UICustomizer can run along side - but there might be conflicts.

Btw. there is planned for color schemes in an upcomming release.

LazeMSS, 2021-11-04:
Okay if we close this?

PRGeno, 2021-11-04:
Yes, of course, my issue is resolved.  Thanks again for your efforts

closed 2021-11-04 by PRGeno