LazeMSS/OctoPrint-UICustomizer #219: Make all of UICustomizer (layouts, themes, etc...) per user or bundles selectable by URL query parameter(s).

bwerst, 2021-11-17:
I've been using UICustomizer for about 6 months or so and really like it.  I've been using my laptop to interface with Cura, Octoprint, and my printer.  I'm hitting the point where I need have a more capable interface right at the printer.  I'd like to do a 10-14" monitor connected directly to the Raspberry Pi running Octoprint.  Immediately, I'm hit with a big UI difference -- on my laptop I like the webcam front and center however, a screen right next to the printer doesn't need the webcam.  Maybe in a "normally closed" panel just to verify its working and aligned correctly, but not taking up valuable default screen real estate.  I don't particularly want to do the tiny screens people are doing with OctoDash, etc....  I'd like to minimize drilling through menus and scrolling.

I'm sure its a ton of work to change how UICustomizer works, but per user or settings bundles seem like the best way to accommodate all sorts of interesting combinations.  At least from an end-user perspective.

Use Cases:
1) Laptop or desktop anywhere, away from the printer.
2) Full interface right at the printer with a different layout.
3) Small interface right at the printer (think OctoDash, TouchUI, etc....)
4) Dashboard Interface - Select panels being used in an IFrame or similar to create a dashboard.  Using multiple frames for multiple instances.  This could be like a security camera type setup for multiple printers or Graphana for stats, etc....  Home Assistant?
5) Phone - Let's user create a different layout that may show/hide different panels relative to use cases 1-4.  For me on my phone all I really need is the webcam, cancel button, and a button to turn on/off the lights in the room.  Using the existing responsive version means lots of scrolling.

I think UICustomizer has all of the features to make great screens for any single use case above.  I think UICustomizer could do all of the use cases if it was linked to the logged in user.  However, not knowing the internals of UICustomizer nor Octoprint maybe it's just Super Themes requested as a url query parameter....

If any of this is actually already possible, can you please point me to any examples.

LazeMSS, 2021-11-17:
Well basically the layout is responsive but you can per device save which panel is closed and open. You need to enable it in settings. But having different layout/profiles is an option but it would require a major rewrite due the way settings are stored - but maybe a bit easier after the latest export/import solution.

LazeMSS, 2021-12-07:
Sorry but this is out of scope of what UI Customizer can do right now - If this was to be implemented is has to be directly in the octoprint - the current way UI Customizer does is it basically just turning on a couple of more things :)

closed 2021-12-07 by LazeMSS