LazeMSS/OctoPrint-UICustomizer #222: Setting all panels in a column to hidden makes the column 0px and panels are now lost.

DrieStone, 2021-12-07:
I hid all of my panels in the 3rd column, and after saving the column width became 0 and I can't make it bigger.

I can drag another panel into the column which allows me to change the column width, but some of my panels are now missing (don't show up at all).

Printer Notifications, Attention, and Octoklipper panel are no longer in my list.

[image]

LazeMSS, 2021-12-07:
Thanks for reporting - will look into it right away

LazeMSS, 2021-12-07:
Until I have it fixes should probally uninstall the plugin and reinstall :)

LazeMSS, 2022-01-03:
Fixed in 0.1.8.1

closed 2022-01-03 by LazeMSS