LazeMSS/OctoPrint-UICustomizer #225: Webcam minimize does not close webcam stream in chromium 96

Tra5is, 2022-01-08:
Each time the webcam maximize button is clicked a new webcam stream is initiated with the server. The stream is not stopped when the view is minimized. 

1. Open the browser's F12 menu and show the network tab.
2. Maximize a webcam stream using the icon in the upper right of the image
3. The network tab will show a stream from the server. 
4. Minimize the webcam view. The stream does not close. It continues to download the stream. 
5. Maximize the webcam stream again.
6. The network tab will show both streams.
7. Continue until your pi, network or browser is out of resources...

Reproduced in 
* Chrome 96.0.4664.110
* Edge 96.0.1054.62

UICustomizer 0.1.8.0

Only about concurrent 3 streams will cause HAProxy to use 100% cpu on one core.

Firefox 95.0.2 doesn't have the same issue. It is able to expand and minimize the webcam stream many times without issue. It doesn't appear to make additional requests to the server for each maximize.

LazeMSS, 2022-01-08:
Thanks will looki ASAP

Tra5is, 2022-01-12:
Is any additional info needed? I can repro easily.

LazeMSS, 2022-01-12:
No info is needed. Haven't had the time to look into it yet.

LazeMSS, 2022-02-07:
This bug is turning out to be quite a serious one - I will be rewriting quite a bit of the main logic

LazeMSS, 2022-02-18:
Fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.8.7

closed 2022-02-18 by LazeMSS