LazeMSS/OctoPrint-UICustomizer #227: 0.1.8.1 update - menu shows only 4 items

cicinovec, 2022-01-15:
After update to latest 0.1.8.1, the menu show only 4 items and the scroll bar has appeared, even on big screen when not necessary
[image]
[image]

Tested in Chrome and Firefox

Arsolix, 2022-01-21:
I have the same problem also.

MrRexRuthless, 2022-01-28:
Same problem here two different installations.

LazeMSS, 2022-01-28:
I'm fixing it today

LazeMSS, 2022-01-30:
@Arsolix @cicinovec @MrRexRuthless could you please post a bug report with you settings or maybe a screenshot - i can not recreate it. Maybe just provide browser and os? But mostly what settings are being used.

MrRexRuthless, 2022-01-30:
Please let me know if this is not what you need.  I'm accessing the PI/OctoPrint from Windows 10 using Edge.  It's happening on two different installations of OctoPrint.  Here are the logs as well.

[octoprint-logs.zip](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/7966692/octoprint-logs.zip)

Here is a screenshot as well

[image]

[image]

This is one shot from each installation.

LazeMSS, 2022-01-30:
Take a screenshot of the settings screen - and maybe use the "report a bug" button - that way i can see other plugins that might cause problems.

MrRexRuthless, 2022-01-30:
[image]
[image]
[image]
[image]

Done (issue 228)

cicinovec, 2022-01-30:
Windows 10, Chrome and Firefox, both the same behavior.

Herebelow "report bug" export

**Plugins installed**
- Autologin Configuration[autologin_config] v0.2.0
- Autoscroll[autoscroll] v0.0.3
- Backup Scheduler[backupscheduler] v0.1.0
- BLTouch Plugin[BLTouch] v0.3.4
- Custom Control Editor[customControl] v0.2.2
- Detailed Progress[detailedprogress] v0.2.7
- DisplayZ[displayz] v0.2.0
- MultiCam[multicam] v0.2.8
- Navbar Temperature Plugin[navbartemp] v0.14
- OctoKlipper[klipper] v0.3.9.4
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- PSU Control[psucontrol] v1.0.6
- PSU Control - Tasmota[psucontrol_tasmota] v1.0.1
- SpoolManager Plugin[SpoolManager] v1.6.0
- **UI Customizer[uicustomizer] v0.1.8.1**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: false
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:96.0) Gecko/20100101 Firefox/96.0

LazeMSS, 2022-01-30:
its a bug in Spoolmanager plugin:
OllisGit/OctoPrint-SpoolManager#230

closed 2022-01-30 by LazeMSS