LazeMSS/OctoPrint-UICustomizer #228: Bug report Issue #227

MrRexRuthless, 2022-01-30:
## Description
**ENTER DESCRIPTION HERE
Describe your problem?
What is the problem?
Can you recreate it?
Did you try disabling plugins?
Did you remember to update the subject?**
<hr>

**Plugins installed**
- BetterHeaterTimeout[BetterHeaterTimeout] v1.3.0
- Creality-2x-temperature-reporting-fix[ender3v2tempfix] v0.0.4
- Detailed Progress[detailedprogress] v0.2.7
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- Exclude Region[excluderegion] v0.3.0
- Marlin EEPROM Editor[eeprom_marlin] v3.2.0
- Navbar Temperature Plugin[navbartemp] v0.14
- OctoDash Companion[octodashcompanion] v0.0.7rc4
- OctoEverywhere![octoeverywhere] v1.5.4
- OctoPod Plugin[octopod] v0.3.12
- Preheat Button[preheat] v0.8.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- SpoolManager Plugin[SpoolManager] v1.6.0
- TP-Link Smartplug[tplinksmartplug] v1.0.1
- **UI Customizer[uicustomizer] v0.1.8.1**

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/97.0.4692.99 Safari/537.36 Edg/97.0.1072.76

LazeMSS, 2022-01-30:
its a bug in Spoolmanager plugin:
https://github.com/OllisGit/OctoPrint-SpoolManager/issues/230

LazeMSS, 2022-01-30:
Until fixed you can add this to custom css under advanced in UICustomizer:
```
.dropdown-menu {
    max-height: none;
    overflow-y: visible;
    overflow-x: visible;
}

#tab_plugin_SpoolManager .dropdown-menu{
    max-height: 120px;
    overflow-y: auto;
    overflow-x: hidden;
}
```

closed 2022-01-30 by LazeMSS