LazeMSS/OctoPrint-UICustomizer #230: [BUG] Touch UI not working

mouse100, 2022-02-08:
Touch UI doesn't work when UICustomizer is installed.
I've tried to reach from my smartphone browser http://myoctoprintip/#touch and touch ui is loaded only for some seconds and then the browser switch to UI Customizer mobile layout. 
I've also tried to disable UI Customizer mobile layout from plugin settings but the issue is still there.

Tried with latest octoptint release 1.7.3

LazeMSS, 2022-02-13:
It works fint here when testing - you need make sure that the hashtag is kept when loading:
ie:
http://ip/#touch 

try using something like this:
http://myoctoprintip/?touch#touch
This will make sure the browser uses another url

LazeMSS, 2022-02-20:
@mouse100 please check this

closed 2022-02-20 by mouse100