LazeMSS/OctoPrint-UICustomizer #231: no scroll/pan/zoom on popup-windows on iOS

Chromogen, 2022-02-10:
When i access octoprint via my iPhone-Browser, i can not pan, scroll or zoom  in or on any popup-window. With bigger windows, there is no option to access the close button, blocking all functionality.

iPhone X
iOS .

Can you recreate it?
Always happens. 100% of the time.

Did you try disabling plugins?
I tried to disable all UI-related plugins but nothing changed.

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.8.1**
- Top Temp[toptemp] v0.0.1.7
- Terminal Messaging[terminalmessaging] v0.1.0
- PrintJobHistory[PrintJobHistory] v1.16.0
- OneDrive Backup[onedrive_backup] v1.0.1
- Translate Model Plugin[translatemodel] v0.2.0
- Telegram Notifications[telegram] v1.6.5
- Printer Stats[stats] v3.1.2
- SpoolManager Plugin[SpoolManager] v1.6.1
- Gcode Leveling[gcodeleveling] v0.4.0
- GcodeEditor[GcodeEditor] v0.2.12
- FileManager[filemanager] v0.1.6
- Exclude Region[excluderegion] v0.3.0
- Bed Cooldown[bedcooldown] v0.4.1
- Octolapse[octolapse] v0.4.1
- Marlin GCode Documentation[marlingcodedocumentation] v0.15.0
- Filament Scale[filament_scale] v0.2.11
- Consolidate Temp Control[consolidate_temp_control] v0.1.8
- Bed Leveling Wizard[bedlevelingwizard] v0.2.4

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi 0.17.0

LazeMSS, 2022-02-11:
What are popups in your mind - is it something like the settings window?
Could you post a screenshot?

Here is how the chrome iphone preview looks with settings open:
[image]

Chromogen, 2022-02-13:
yep, that's what i ment. Popups or popouts? or whatever they are called. :)
With windows of that size, there are no problems as you do not need to scroll/pan, but with larger ones, for example print-job-history-report, the buttons are out of screen and no chance to move the view at all.

On my device, when i try to scroll down to close/save buttons, safari pops the window to the overview screen/safari tab view. So i guess it does not recognize any scrollable window. Thus you are stuck with that view and damned to close and reopen the tab and hope, the window wont show up again.

I did a bit of research and it seems to be a common problem with iOS on many websites. So one can consider it a bug in iOS? Anyhow, it works for some reason without UICustomizer :/

[image]

LazeMSS, 2022-02-13:
Yes really not a UI Customizer problem really - its the way the old bootstrap works (when mobile/responsive mode) is turned on and then other plugins don't considered responsive screens. I'm hoping this does not happen in any of the "native" modals/popovers like settings.

I can try and make a "native" patch for all popovers

Chromogen, 2022-02-13:
thank you kindly, will test asap-rinter is done with current job :)

this is pretty much the reason why i always avoided to create / mess with any UI stuff. i am so very thankful that people like you deliberately jump into that breach...

Chromogen, 2022-02-14:
aaaaand here we go: print finished ans all set up.

and it works like a charm! my man! thank you so much! <3

LazeMSS, 2022-02-14:
Will close when released

LazeMSS, 2022-02-18:
Fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.8.7