LazeMSS/OctoPrint-UICustomizer #235: Webcam view closing

NeilSM, 2022-02-20:
## Description
**ENTER DESCRIPTION HERE
Describe your problem?
- Webcam view closing
What is the problem?
- On tabs other than the main tab the web cam view closes after a few seconds and can not be reopened unless you navigate back to the main tab
Can you recreate it?
- Yes
Did you try disabling plugins?
- No
Did you remember to update the subject?**
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.8.7**
- TP-Link Smartplug[tplinksmartplug] v1.0.1
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- Resource Monitor[resource_monitor] v0.3.6
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- PrettyGCode[prettygcode] v1.2.4
- Floating Navbar[floatingnavbar] v0.3.7
- Exclude Region[excluderegion] v0.3.0
- Calibration Tools[CalibrationTools] v1.0.4
- BLTouch Plugin[BLTouch] v0.3.4
- Octolapse[octolapse] v0.4.1
- OctoApp[octoapp] v1.0.8
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- Change Filament Plugin[Change_Filament] v0.3.2
- Bed Visualizer[bedlevelvisualizer] v1.1.1

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: true
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/98.0.4758.102 Safari/537.36 Edg/98.0.1108.56

LazeMSS, 2022-02-20:
Whats the URL of you webcam stream?

Can you post a screenshot of how it looks when the web cam view has been closed?

NeilSM, 2022-02-20:
The webcam doesn't disappear immediately, it's stays visible for 10 to 20 seconds.  

[image]

NeilSM, 2022-02-20:
the stream URL is
/webcam/?action=stream

NeilSM, 2022-02-20:
Also getting camera view duplication when swapping tabs:
[image]

LazeMSS, 2022-02-20:
Wauw. 
Do you have any custom CSS entered?

There must be something confusing my plugin. This should never happen. 🙈

NeilSM, 2022-02-20:
Hi, no custom css, just the plugins listed. It started happening right after the last update.

LazeMSS, 2022-02-20:
Yes but making a second copy should not be possible because it basically moves the webcam around.

Could you try and refresh your browsers cache by pressing shift+f5 but it should not happen.

CBDesignS, 2022-02-21:
> Yes but making a second copy should not be possible because it basically moves the webcam around.
> 
> Could you try and refresh your browsers cache by pressing shift+f5 but it should not happen.
> 
> Or if you know how could you use the developer console in the browser to inspect the dom for the webcam widget

I can second the double stream. It seems to happened if float screen (the work arround you mentioned in my bug report) and the stream is active in control tab then switch to a different tab it sometimes duplicates the stream window
[image]
.

CBDesignS, 2022-02-21:
not sure what info you want from browser debug.. 
the next img is after selecting the top webcam stream panel
[image]
the next img is after selecting the 2nd bottom stream panel.
[image]

Not sure if you can zoom into the images and read the contect of the debug info.

NeilSM, 2022-02-22:
After updating my other octoprint server on my CR10 I get the same behaviour with a small difference, I get the "camera loading" message when it swap tabs.
[image]

NeilSM, 2022-02-22:
Update: the webcam sometimes will reload if you just stay on the tab, it can take a little while but it does come back. If you navigate back to the main tab it will also sometimes just reappear.

LazeMSS, 2022-02-24:
@NeilSM  and @CBDesignS  having "Floating Navbar" installed could be a conflict but its not related to this issue.

LazeMSS, 2022-02-24:
This is a duplicate of #234 but keeping it open. I'm 99% sure the bug is related to prettygcode - I have submitted a bug report and a PR to fix it for that plugin.

Until then you can use my the PR version pretty gcode by manual installing the plugin from: https://github.com/LazeMSS/OctoPrint-PrettyGCode/archive/refs/heads/id-bugfix.zip
(for help see: https://plugins.octoprint.org/help/installation/ under "Via the URL and the Plugin Manager")

NeilSM, 2022-02-24:
> This is a duplicate of #234 but keeping it open. I'm 99% sure the bug is related to prettygcode - I have submitted a bug report and a PR to fix it for that plugin ([Kragrathea/OctoPrint-PrettyGCode#135](https://github.com/Kragrathea/OctoPrint-PrettyGCode/pull/135))
> 
> Until then you can use my the PR version pretty gcode by manual installing the plugin from: https://github.com/LazeMSS/OctoPrint-PrettyGCode/archive/refs/heads/id-bugfix.zip (for help see: https://plugins.octoprint.org/help/installation/ under "Via the URL and the Plugin Manager")

Brilliant, thanks for looking into this :)
Not got the printers running today but will test it out as soon as I get a minute.
N.

NeilSM, 2022-02-24:
Disabled PrettyGcode and tested:
- swapping tabs
- Full screen cam and back to window
- Close / Open webcam
All worked fine no random loss of camera feed.

Installed code from your link, tested same as above and all working fine.
Thank you.

LazeMSS, 2022-03-05:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.8.9
if not then please reopen this issue.

closed 2022-03-05 by LazeMSS