LazeMSS/OctoPrint-UICustomizer #236: Floating UI Camera

JesNoFear, 2022-02-22:
when set to floating UI camera, the text that shows at the bottom of the box covers over the bottom half of the video feed, instead of appending to the bottom of the feed. not a huge deal, but it would be nice to see all that is in camera view. if you try and stretch it, you can see full feed but it wont stay that size, pops back to the original size.

<hr>

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Change Filament Plugin[Change_Filament] v0.3.2
- Cura Thumbnails[UltimakerFormatPackage] v1.0.1
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- Google Drive Backup[googledrivebackup] v0.1.0
- Long-SD-list-paths[LongPathSDList] v1.1.0
- Marlin EEPROM Editor[eeprom_marlin] v3.2.0
- MyMiniFactory[myminifactory] v0.1.4
- Octoprint Display ETA[display_eta] v2.1.0
- OneDrive Backup[onedrive_backup] v1.0.1
- PrettyGCode[prettygcode] v1.2.4
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- Slicer Estimator[SlicerEstimator] v1.3.1
- SpoolManager Plugin[SpoolManager] v1.6.1
- Sticky Pad[stickypad] v0.1.4
- TimeToFilament Plugin[TimeToFilament] v1.2.1
- Top Temp[toptemp] v0.0.1.7
- **UI Customizer[uicustomizer] v0.1.8.7**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: true
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:97.0) Gecko/20100101 Firefox/97.0

LazeMSS, 2022-02-24:
Fixed in next release.

LazeMSS, 2022-03-05:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.8.9
if not then please reopen this issue.

closed 2022-03-05 by LazeMSS