LazeMSS/OctoPrint-UICustomizer #237: Camera Wont show in layout

Mr-Vulcan, 2022-02-26:
## Description
**ENTER DESCRIPTION HERE
Describe your problem?
Open the system and the camera wont show on the camera layout unit however it will show in dashboard plugin which is integrated with the ui customizer 

**Plugins installed**
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f
- Backup Scheduler[backupscheduler] v0.1.0
- BetterHeaterTimeout[BetterHeaterTimeout] v1.3.0
- Cost Estimation[costestimation] v3.5.0
- Dashboard[dashboard] v1.19.8
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- FileManager[filemanager] v0.1.6
- Firmware Updater[firmwareupdater] v1.13.1
- Preheat Button[preheat] v0.8.0
- PrettyGCode[prettygcode] v1.2.4
- Printer Stats[stats] v3.1.2
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- Resource Monitor[resource_monitor] v0.3.6
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- Smart Preheat[smartpreheat] v0.0.6
- Telegram Notifications[telegram] v1.6.5
- **UI Customizer[uicustomizer] v0.1.8.7**

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: false
- compactMenu: false
- compressTempControls: false
- disableTermInactive: false
- filesFullHeight: false
- fixedFooter: false
- fixedHeader: false
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: false
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/98.0.4758.102 Safari/537.36

LazeMSS, 2022-02-26:
Try disabling pretty gcode plugin? Is the problem still there?

Mr-Vulcan, 2022-02-26:
> Try disabling pretty gcode plugin? Is the problem still there?

Thanks for the idea. That hasent helped.

LazeMSS, 2022-02-26:
Could try and explain a bit more. 
What is "camera layout unit"? Could you post a screenshot?

I don't have any integration with dashboard plugin but I will test for know problems. 

There is another open issue regarding webcam problems you could also look if your problems sounds familiar.

Mr-Vulcan, 2022-02-26:
> > Could try and explain a bit more. What is "camera layout unit"? Could you post a screenshot?
> > I don't have any integration with dashboard plugin but I will test for know problems.
> > There is another open issue regarding webcam problems you could also look if your problems sounds familiar.
> 
> Sure [image] As you can see in the snapshot the camera is visible on the center collum. That layout on the center collum is using another plugin called Dashboard which is displaying the camera

I've just been able to fix it. All I did was enable float/fullscreen webcam and select Fullscreen Webcam
Cheers for the help!

LazeMSS, 2022-02-27:
Will be fixed in next released

LazeMSS, 2022-03-05:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.8.9
if not then please reopen this issue.

closed 2022-03-05 by LazeMSS