LazeMSS/OctoPrint-UICustomizer #238: [BUG] Camera feed goes away after zoom if PrintJobHistory plugin is enabled

ManuelMcLure, 2022-03-01:
## Description
This is the problem I originally reported in #234 - opening a new issue as requested.
If the PrintJobHistory plugin is installed and enabled, zooming and unzooming the camera makes the camera feed go away (images posted in #234).
If I disable PrintJobHistory then the problem does not occur. This happens even if the only third-party plugins installed are UI Customizer and PrintJobHistory.
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.8.7**
- PrintJobHistory[PrintJobHistory] v1.16.0

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: false
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: true
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: false
- saveAccordions: true
- themeLocal: false

**Software versions**
- OctoPrint 1.7.3
- Python 3.9.2

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/98.0.4758.102 Safari/537.36
- Also happens on Chrome for Mac OS and Edge

LazeMSS, 2022-03-01:
Awesome thanks

ManuelMcLure, 2022-03-03:
Initial test with your new branch looks good (Chrome and Edge on MacOS).

LazeMSS, 2022-03-05:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.8.9
if not then please reopen this issue.

closed 2022-03-05 by LazeMSS

ManuelMcLure, 2022-03-06:
Can confirm 0.1.8.9 fixes the problem.