LazeMSS/OctoPrint-UICustomizer #239: Bug: Multiple instances of Octoprint on one Pi each with a different theme

Canabl3d, 2022-03-01:
## Description
**ENTER DESCRIPTION HERE
Describe your problem?  I have three instances of Octoprint installed on one raspberry pi, one for each of my 3D printers. I have applied different coloured themes to each printer to make it readily identifiable. When I do an initial load in chrome or Microsoft edge all the printers load up with the proper coloured theme assigned. Unfortunately if I refresh on any tab the theme gets updated with the last one that was loaded regardless which printer it is assigned to.
Example:  Initial load in this order   printer one = red    printer two = blue   printer three = green
now if I hit refresh on printer one or printer two the theme will be updated to green which is assigned to printer three. In the settings for the UI customizer the theme selected will still show the proper one assigned to the printer even though it is using the theme from the last one loaded. As an example in the settings printer one still thinks it has the red theme loaded. The only way to refresh the theme properly is to restart that instance of octoprint or go into the settings and change the theme to something different and then back to the red.  Its like ui customizer loses track of which theme variables apply to which open browser tab.

What is the problem?  see above
Can you recreate it?    easy to recreate following steps above
Did you try disabling plugins?   yes, even on a clean install with no plugins  
Did you remember to update the subject?**  yes
<hr>

**Plugins installed**
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f
- Autoscroll[autoscroll] v0.0.3
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Calibration Tools[CalibrationTools] v1.0.4
- Cancel Objects[cancelobject] v0.4.7
- Consolidate Temp Control[consolidate_temp_control] v0.1.8
- Dashboard[dashboard] v1.19.8
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- Draggable Files[draggable_files] v1.1.2
- Exclude Region[excluderegion] v0.3.0
- FileManager[filemanager] v0.1.6
- Firmware Updater[firmwareupdater] v1.13.1
- Floating Navbar[floatingnavbar] v0.3.7
- GcodeEditor[GcodeEditor] v0.2.12
- HeaterTimeout[HeaterTimeout] v0.0.4
- Navbar Temperature Plugin[navbartemp] v0.14
- Octolapse[octolapse] v0.4.1
- Preheat Button[preheat] v0.8.0
- PrettyGCode[prettygcode] v1.2.4
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- Resource Monitor[resource_monitor] v0.3.6
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- UBL Mesh Editor[ublmeshedit] v0.3.2
- **UI Customizer[uicustomizer] v0.1.8.7**
- ΔµCalibrator[DuCalibrator] v1.0.4

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: false
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/98.0.4758.102 Safari/537.36

Canabl3d, 2022-03-02:
> How are the multiple instances installed on the octopi?

I followed the tutorial in this YouTube video so that I could run all 3 printers on one Pi
https://www.youtube.com/watch?v=FIPbiImixZM&t=1346s&ab_channel=ChrisRiley

By the way, I love your plugin, amazing work!  Thank you

LazeMSS, 2022-03-02:
It might be related to how the CSS is included - I will try and do a double install following that guide later.

Canabl3d, 2022-03-02:
That would be great mate. Let me know if you need anything more from me along the way.

LazeMSS, 2022-12-28:
https://community.octoprint.org/t/multiple-printers-per-server-megathread/7778/112

LazeMSS, 2023-03-05:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.1 - please reopen if not working.

closed 2023-03-05 by LazeMSS