LazeMSS/OctoPrint-UICustomizer #241: Loosing camera after fullscreen

SpectreGadgetReborn, 2022-03-06:
I frequently go into full screen camera view to see better detail.  
When I exit full-screen mode, i no longer have the camera feed, and that spot is just blinks, "Webcam stream loading...".  If I click on another tab, the smaller webcam section just blinks, "Loading webcam..."
Reloading the webpage brings things back to normal.
In testing, one time I exited full screen, and it was showing a frozen frame from the webcam, instead of it disappearing.  I could not get it to show a live view (or switch cameras) until reloading the page.

**Plugins installed**
- WiFi Status[wifistatus] v1.7.2
- **UI Customizer[uicustomizer] v0.1.8.9**
- Sticky Pad[stickypad] v0.1.4
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- Octoslack[Octoslack] v2.2.0
- Cura Thumbnails[UltimakerFormatPackage] v1.0.1
- TP-Link Smartplug[tplinksmartplug] v1.0.1
- Tempsgraph Plugin[tempsgraph] v0.3.7
- RequestSpinner[requestspinner] v0.2.0
- Prusa Leveling Guide[PrusaLevelingGuide] v1.0.17
- MultiCam[multicam] v0.3.0
- MQTT[mqtt] v0.8.12
- ipOnConnect[ipOnConnect] v0.2.4
- Firmware Updater[firmwareupdater] v1.13.1
- Dragon Order[dragon_order] v0.1.7
- Draggable Files[draggable_files] v1.1.2
- DisplayProgress[displayprogress] v0.1.3
- Dashboard[dashboard] v1.19.8
- Custom Control Editor[customControl] v0.2.2
- Auto Terminal Input[AutoTerminalInput] v0.1.0
- Octolapse[octolapse] v0.4.1
- OctoDash Companion[octodashcompanion] v0.0.6
- M73 ETA Override[m73etaoverride] v1.0.4
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- Bed Visualizer[bedlevelvisualizer] v1.1.1

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: false
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: true
- themeLocal: false

**Software versions**
- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:97.0) Gecko/20100101 Firefox/97.0

LazeMSS, 2022-03-06:
Im unable to reproduce this on my Windows machines running firefox - can you test if the problem is there on Chrome - in general it seems like webcams streams in firefox run wild

SpectreGadgetReborn, 2022-03-06:
I installed Chrome and it produces the same result.  
I also tried Safari; it opens a fullscreen view and immediately closes it.  The main window produces the same result.

LazeMSS, 2022-03-06:
Strange could you try and disabling multi cam plugin?

SpectreGadgetReborn, 2022-03-07:
With the multicam plugin disabled, exiting full screen camera restores the regular camera view correctly in Firefox.
In Safari, it tries to go into 'proper' MacOS full screen, but still immediately exits it.  The camera view is correctly restored; however.

LazeMSS, 2022-03-11:
Its a know problem with multicam plugin so I will be fixed in the next release of this.

closed 2022-03-11 by LazeMSS