LazeMSS/OctoPrint-UICustomizer #242: [BUG] Window width layout and window styles went nuts

elpeter25, 2022-03-11:
## Description
Since version 0.1.8.7 page layout become nuts and so the icons and styles. Page width seems to have scroll as UI Customizer adds some margins left and right. Also control icons for temperature and so on went bigger making the style of the temp controls weird.
Please see pics so you can find differences I tried to describe.

Describe your problem? Page styles issue
What is the problem? Page and icons size are wrong.
Can you recreate it? Yes. I have screenshots of "before" and "after" the upgrade, also have an export file of 0.1.8.1 that I've used to revert back to make it work again after downgrading to that version. Will also attach it.
Did you try disabling plugins? Yes tried to disable as many plugins as I thought could clash but none fix it. Only downgrading and importing previous settings worked.

<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.8.9**
- TouchUI[touchui] v0.3.18
- Top Temp[toptemp] v0.0.1.7
- Terminal Messaging[terminalmessaging] v0.1.0
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- Simple Filament Change Buttons[simple_filament_change_buttons] v1.1
- Print Scheduler[printscheduler] v0.0.4
- TimeToFilament Plugin[TimeToFilament] v1.2.1
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- Navbar Temperature Plugin[navbartemp] v0.14
- InlineConfirm[inlineconfirm] v1.1.4
- HeaterTimeout[HeaterTimeout] v0.0.4
- GcodeEditor[GcodeEditor] v0.2.12
- Floating Navbar[floatingnavbar] v0.3.7
- DisplayProgress[displayprogress] v0.1.3
- Custom Control Editor[customControl] v0.2.2
- Cancel Objects[cancelobject] v0.4.7
- Auto Terminal Input[AutoTerminalInput] v0.1.0
- Autoscroll[autoscroll] v0.0.3
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- Consolidate Temp Control[consolidate_temp_control] v0.1.8
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Backup Scheduler[backupscheduler] v0.1.0

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.7.3
- Python 3.6.9

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:98.0) Gecko/20100101 Firefox/98.0

[image]
[image]
[UICustomizerSettings.json.txt](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/8231827/UICustomizerSettings.json.txt)

LazeMSS, 2022-03-23:
Strange is this still a problem?

elpeter25, 2022-03-23:
> Strange is this still a problem?

Yep, I'm afraid so. I have to install version 1.8.1 to have a proper view, otherwise it seems like the centre column has a wide right margin... I noticed it also on terminal tab.
Yesterday I installed two fresh octoprint installations and restoring the backup I had installed the latest version of the plugin so to make it visible I had to uninstall and install manually version 1.8.1...

LazeMSS, 2022-03-24:
Can you take a screenshot of the layout tab in UI customizer settings?

elpeter25, 2022-03-24:
> Can you take a screenshot of the layout tab in UI customizer settings?
[image]

LazeMSS, 2022-03-29:
Sorry for the slow response - does it happen with all themes?

LazeMSS, 2022-05-06:
@elpeter25 ?

elpeter25, 2022-05-19:
> @elpeter25 ?

Sorry. I'm back.
Yes I've tried also with other themes and also happens.
Also I just updated to 1.8.0 of OP and still happens. Here you go the screenshot with other theme on latest OP.
[image]

LazeMSS, 2022-05-21:
Can you try disabling Consolidate Temp Contro

LazeMSS, 2022-05-23:
Ohh and btw. no need to have both Navbar Temperature Plugin and TopTemp. And also the Floating Navbar is also included in UI Customizer - that might also be causing an conflict.

The size of the icons/buttons in the temperature can be controlled with the "Compact" option in UI Customizer but it might collide with the many other plugins you are running.

elpeter25, 2022-05-23:
> Can you try disabling Consolidate Temp Control

That made the trick!
So it is a compatibility issue with that plugin then... but it didn't happen up to version 1.8.1

elpeter25, 2022-05-23:
> Ohh and btw. no need to have both Navbar Temperature Plugin and TopTemp. And also the Floating Navbar is also included in UI Customizer - that might also be causing an conflict.
> 
> The size of the icons/buttons in the temperature can be controlled with the "Compact" option in UI Customizer but it might collide with the many other plugins you are running.
> 
> I would recommend you try turning off a plugin one at the time to figure out which one is causing the problems I have tried installing a couple of them but could not recreate it.

Thanks! I've disabled Navbar Temperature Plugin and Floating Navbar :)

LazeMSS, 2022-05-25:
> > Can you try disabling Consolidate Temp Control
> 
> That made the trick! So it is a compatibility issue with that plugin then... but it didn't happen up to version 1.8.1

I wonder why that happend - I can't recreate :/

elpeter25, 2022-05-25:
That's odd... for me disabling the consolidate temp control worked but I neeeeeed that plugin
Find attached the exported list of my installed plugins just in case it might help...
Not attached as it doesn't allow me to upload json so I tweaked it to be readable here:

```
"key": "uicustomizer", "name": "UI Customizer", "url": "https://github.com/LazeMSS/OctoPrint-UICustomizer"
"key": "touchui", "name": "TouchUI", "url": "https://github.com/BillyBlaze/OctoPrint-TouchUI"
"key": "toptemp", "name": "Top Temp", "url": "https://github.com/LazeMSS/OctoPrint-TopTemp"
"key": "terminalmessaging", "name": "Terminal Messaging", "url": "https://github.com/jeffeb3/OctoPrint-TerminalMessaging"
"key": "prusaslicerthumbnails", "name": "Slicer Thumbnails", "url": "https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails"
"key": "simple_filament_change_buttons", "name": "Simple Filament Change Buttons", "url": "https://github.com/TheThief/Octoprint_Simple_Filament_Change_Buttons"
"key": "printscheduler", "name": "Print Scheduler", "url": "https://github.com/jneilliii/OctoPrint-PrintScheduler"
"key": "TimeToFilament", "name": "TimeToFilament Plugin", "url": "https://github.com/eyal0/OctoPrint-TimeToFilament"
"key": "simpleemergencystop", "name": "Simple Emergency Stop", "url": "https://github.com/Sebclem/OctoPrint-SimpleEmergencyStop"
"key": "PrintTimeGenius", "name": "PrintTimeGenius Plugin", "url": "https://github.com/eyal0/OctoPrint-PrintTimeGenius"
"key": "inlineconfirm", "name": "InlineConfirm", "url": "https://github.com/j7126/OctoPrint-InlineConfirm"
"key": "HeaterTimeout", "name": "HeaterTimeout", "url": "https://github.com/Andy-ch/OctoPrint-HeaterTimeout"
"key": "GcodeEditor", "name": "GcodeEditor", "url": "https://github.com/ieatacid/OctoPrint-GcodeEditor"
"key": "displayprogress", "name": "DisplayProgress", "url": "https://github.com/OctoPrint/OctoPrint-DisplayProgress"
"key": "customControl", "name": "Custom Control Editor", "url": "https://github.com/Salandora/octoprint-customControl"
"key": "cancelobject", "name": "Cancel Objects", "url": "https://github.com/paukstelis/OctoPrint-Cancelobject"
"key": "DisplayLayerProgress", "name": "DisplayLayerProgress Plugin", "url": "https://github.com/OllisGit/OctoPrint-DisplayLayerProgress"
"key": "consolidate_temp_control", "name": "Consolidate Temp Control", "url": "https://github.com/jneilliii/OctoPrint-ConsolidateTempControl"
"key": "bedlevelvisualizer", "name": "Bed Visualizer", "url": "https://github.com/jneilliii/OctoPrint-BedLevelVisualizer"
"key": "backupscheduler", "name": "Backup Scheduler", "url": "https://github.com/jneilliii/OctoPrint-BackupScheduler"
```

Thanks for your time!!

elpeter25, 2022-05-25:
IDK if this can be of any help...
[image]
[image]
[image]

elpeter25, 2022-05-30:
FYI I think I found something... In the "Consolidate Temp Control" plugin I have it set to "horizontal view" in settings.
[image]

If I change it to vertical there's no issue:
[image]

LazeMSS, 2022-06-17:
@elpeter25  sorry for the long delay - I just tested - i now have the same problem but as soon as there is data in the terminal it is fixed.
I noticed in all you screenshots there is no data in the terminal. Can you test with data/connect to printer - does the terminal expand

LazeMSS, 2022-06-17:
Found the bug and working with @jneilliii  to fix it :)

elpeter25, 2022-06-17:
> Found the bug and working with @jneilliii to fix it :)

Awesome!! Thanks so so much for your work!! :)

jneilliii, 2022-06-17:
Two options here. Either use my other Consolidated Tabs plugin or I just patched and TempControl you can install by copying/pasting the URL into Plugin Manager > Get More > ...from URL and click Install.

```
https://github.com/jneilliii/OctoPrint-ConsolidateTempControl/archive/refs/heads/bugfix/uicustomizer.zip
```

[image]

LazeMSS, 2022-06-17:
Closing for now

jneilliii, 2022-06-17:
actually went ahead and created 0.1.9 release of Consolidated Temp Control plugin just now....

elpeter25, 2022-06-17:
> Two options here. Either use my other Consolidated Tabs plugin or I just patched TempControl you can install by copying/pasting the URL into Plugin Manager > Get More > ...from URL and click Install.
> 
> ```
> https://github.com/jneilliii/OctoPrint-ConsolidateTempControl/archive/refs/heads/bugfix/uicustomizer.zip
> ```
> 
> [image]

Hi,

Sorry to trouble you this much but I cannot make it work... I tried installing version you sent above and latest 0.1.9 version from github (as it says been modified just few minutes ago)
Been strugeling with that trying to make it work, unisntalling, cleaning up data, reinstalling octoprint, trying another instance of octoprint, etc none of that make it work for me :(

How could we debug this? Maybe an Anydesk session will help?

Thanks for your time!!

jneilliii, 2022-06-17:
What are your main settings?

[image]

elpeter25, 2022-06-17:
One more test. I disabled every plugin but UI customizer + consolidate and tried another theme just in case:

[image]

elpeter25, 2022-06-17:
> What are your main settings?
> 
> [image]

[image]

jneilliii, 2022-06-17:
That's strange, you have compress temp controls turned on, but they're not compressing. 

[image]

do you have anything added in UI Customizer's Advanced CSS section?

jneilliii, 2022-06-17:
noticed your layout was 3 - 6 - 3, so changed mine to that and get the temp stuff cut off if not compressed, but with compressed option enabled it still shows ok.

[image]

elpeter25, 2022-06-17:
I tried with your settings with no result. Even tried 3 - 7 - 2, no luck either.
Just tested Firefox private window, same. Chrome, same :'(

elpeter25, 2022-06-17:
Brand new installation with only UI Customizer and Consolodate Temp plugins installed. Same behavior:

[image]
[image]

elpeter25, 2022-06-17:
BTW window is 1920x1080 (have a 43" 4k display split in 4)

[image]
[image]

elpeter25, 2022-06-17:
Disabling following rule helps:

http://192.168.0.24:5009/plugin/uicustomizer/static/css/bootstrap-responsive.css
.row-fluid .span10 {
	width: 82.90598290598291%;
	*width: 82.8527914166212%;
}

[image]

elpeter25, 2022-06-17:
And setting this to 12.59 like left column (instead of 14.59) makes it look ok I think...

.row-fluid .span2 {
	width: 12.52991452991453%;
	*width: 14.476723040552828%;
}

[image]

LazeMSS, 2022-06-18:
Will look more later.

LazeMSS, 2022-06-18:
@elpeter25  could you please try and connect to a printer ?

LazeMSS, 2022-06-18:
Something is really broken in your setup / the clean install did you restore settings for UICustomizer or something?
Could try uploading your settingsfile again - looking at your screenshots
https://user-images.githubusercontent.com/37652763/174400030-f1dd0846-15c0-44bf-9a3b-282454cc265a.png
https://user-images.githubusercontent.com/37652763/174399527-a5eb0a02-3462-4616-8100-dd9c816fcdb1.png

There is something setting a span10 on "tabable"
[image]

jneilliii, 2022-07-06:
The changes to my plugin can be seen [here](https://github.com/jneilliii/OctoPrint-ConsolidateTempControl/commit/7ebdec69f24e6b44dff9e552c7c7bfd11146ad48), which is UI Customizer is detected the span10 is not added.

jneilliii, 2022-07-06:
I would personally recommend switching from that Temp Control plugin to my Consolidated Tabs plugin. It should work for you way better with UI Customizer.

LazeMSS, 2022-12-29:
Closing this - I spent 3 hours trying to recreate - please reopen and provide details on how to recreate again

closed 2022-12-29 by LazeMSS

elpeter25, 2023-01-03:
Hi,
First I want to apologize for the late late response. Had some personal problem.

I "solved" it using your suggestions and I now use Consolidated Tabs plugin. Thanks so much for everything!!