LazeMSS/OctoPrint-UICustomizer #243: [Bug] Option "Hide webcamera on Control tab" does not work

Nedial, 2022-03-20:
## Description
**ENTER DESCRIPTION HERE
After updating to the current version, the webcam in the control tab is no longer hidden despite the option being enabled. Deactivating and activating the option did not improve the situation either.
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.8.9**
- Top Temp[toptemp] v0.0.1.7
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- Sidebar Webcam[webcamSB] v0.1.8
- Resource Monitor[resource_monitor] v0.3.6
- Prusa Leveling Guide[PrusaLevelingGuide] v1.0.17
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- GcodeEditor[GcodeEditor] v0.2.12
- Calibration Tools[CalibrationTools] v1.0.4
- Active Filters Extended[active_filters_extended] v0.1.0
- Octolapse[octolapse] v0.4.1
- Marlin GCode Documentation[marlingcodedocumentation] v0.16.0
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Arc-Welder[arc_welder] v1.1.0rc4

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: false
- disableTermInactive: true
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: true
- saveAccordions: false
- themeLocal: true

**Software versions**
- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/99.0.4844.74 Safari/537.36 Edg/99.0.1150.46

LazeMSS, 2022-03-21:
Did you try reloading the page after changing the settings - i can see there might be some random fluke here.

LazeMSS, 2022-04-14:
Hello?

LazeMSS, 2022-05-06:
@Nedial ?

LazeMSS, 2022-05-21:
Closing - no response and unable to reproduce

closed 2022-05-21 by LazeMSS