LazeMSS/OctoPrint-UICustomizer #244: [Feature Request] Respect webcam settings in sidebar view

dnlm, 2022-04-01:
## Description
The way my camera is mounted, I need to check all three rotation options to get a correctly orientated video feed:
[image]

The main camera view in the controls tab still works as expected, but the sidebar view shows the stream 'as-is' (90 degrees clockwise): [image]

It would be great if the sidebar camera view would take the orientation settings into account.

I did not disable any plugins, as I don't have any plugins installed related to the camera settings.

Last but not least: **Thank you** for the great plugin! It's all I wished for and more after using OctoPrint mainly on my phone (via OctoApp) because I don't really like the default web ui.
<hr>

**Plugins installed**
- Cura Thumbnails[UltimakerFormatPackage] v1.0.1
- Filament Manager[filamentmanager] v1.9.1
- MQTT[mqtt] v0.8.12
- OctoApp[octoapp] v1.0.8
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- PSU Control[psucontrol] v1.0.6
- PSU Control - MQTT[psucontrol_mqtt] v1.0.2
- **UI Customizer[uicustomizer] v0.1.8.9**
- Webhooks[webhooks] v3.0.3

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: false
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: true
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: true
- themeLocal: false

**Software versions**
- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/99.0.4844.84 Safari/537.36

LazeMSS, 2022-04-01:
Strange it should do that. Will look again

LazeMSS, 2022-04-01:
Fixed in next release

dnlm, 2022-04-01:
Wow, blazing fast, thank you! Can confirm that did the trick:
[image]

LazeMSS, 2022-04-01:
Please con't close until release - im guessing you installed the develop branch?

dnlm, 2022-04-05:
Sorry, should have known better but it was late Nd the close with comment button looked just right ;)

I saw that the fix was just a missing class so I added it manually.

LazeMSS, 2022-05-25:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.0

Please reopen and comment if not working as expected.

closed 2022-05-25 by LazeMSS