LazeMSS/OctoPrint-UICustomizer #246: Turn off camera stream if camera is not showing

AssisrMatheus, 2022-05-01:
Unlike the default octoprint UI, the webcam stream is not turned off when you're not viewing the webcam..

I'm using octo4a and the phone gets hot when the camera is on for a while (even with fps limiting). Using the default octoprint ui, when you're not actually viewing the webcam. It turns the stream off. And then turns back on if it was off when you want to visualize it again.

But with ui customizer. It seems that even though we can "collapse" the webcam, it's only visual, and the webcam stream is still on in the background, instead of being turned off

LazeMSS, 2022-05-21:
@AssisrMatheus ?

LazeMSS, 2022-05-25:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.0

Please reopen and comment if not working as expected.

closed 2022-05-25 by LazeMSS