LazeMSS/OctoPrint-UICustomizer #247: Hide Webcam in Control not working

linuxdudeg, 2022-05-08:
## Description
I cannot hide the web cam in the control tab
I have tried selecting / de-selecting, also changed floating to see if this affects it, but the webcam is always visible in the control tab
I only have the Dashboard and UI Customizer enabled, still the same

<hr>

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Creality-2x-temperature-reporting-fix[ender3v2tempfix] v0.0.4
- Dashboard[dashboard] v1.19.8
- Firmware Updater[firmwareupdater] v1.13.2
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- **UI Customizer[uicustomizer] v0.1.8.9**

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: true

**Software versions**
- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/101.0.4951.54 Safari/537.36

LazeMSS, 2022-05-12:
I hope to have it fixed in the next release...

LazeMSS, 2022-05-25:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.0

Please reopen and comment if not working as expected.

PR #251 (2022-05-25): 0.1.9.0 - ready
Fixes:
#247
#246
#250
#244

Improved icon search and colors
closed 2022-05-25 by LazeMSS