LazeMSS/OctoPrint-UICustomizer #248: Gcode full-size doesn't work on wide monitor

luizbgomide, 2022-05-11:
## Description
I'm trying to make the gcode viewer use the full width available, but it only uses half the available Area:
[image]

I've tried to tweak every possible setting without luck.

<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.8.9**
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- Preheat Button[preheat] v0.8.0
- Cura Thumbnails[UltimakerFormatPackage] v1.0.2
- Telegram Notifications[telegram] v1.6.5
- SpoolManager Plugin[SpoolManager] v1.7.0
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- Resource Monitor[resource_monitor] v0.3.6
- PSU Control[psucontrol] v1.0.6
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- Printoid Plugin[printoid] v1.2.0
- PrettyGCode[prettygcode] v1.2.4
- PortLister[portlister] v0.1.10
- Navbar Temperature Plugin[navbartemp] v0.14
- MeatPack[meatpack] v1.5.21
- GcodeEditor[GcodeEditor] v0.2.12
- Exclude Region[excluderegion] v0.3.0
- Enclosure Plugin[enclosure] v4.13.2
- Detailed Progress[detailedprogress] v0.2.7
- Dashboard[dashboard] v1.19.8
- Camera Settings[camerasettings] v0.3.2
- Octolapse[octolapse] v0.4.1
- OctoEverywhere![octoeverywhere] v1.6.6
- Marlin GCode Documentation[marlingcodedocumentation] v0.16.0
- GPIO-Status[gpiostatus] v0.2.1
- Google Drive Backup[googledrivebackup] v0.1.1
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Arc-Welder[arc_welder] v1.1.0rc4
- Action Commands[actioncommands] v0.5

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: false
- disableTermInactive: true
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: true
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/101.0.4951.54 Safari/537.36

LazeMSS, 2022-05-12:
Strange - what screen resolution?

I can't recreate - it could be caused by another plugin changing things on the gcode viewer, maybe try disabling a couple of the other plugins

luizbgomide, 2022-05-13:
**Exclude Region** is the culprit. Not sure what it is doing to the gcode window, or if that can be fixed somehow...

I have used it in the past and I think it has its use, unfortunetelly it looks like that is was abandoned 2 years ago.

LazeMSS, 2022-05-13:
I will try and make a workaround in the next release

LazeMSS, 2022-05-22:
Sadly it seems like the Exclude Region plugin set the size of the gcode viewer to a fixed size (in pixels) and thats breaking alot of things. So for now you will need to either have the gcode in "standard" size or disabled the exclude region plugin.

closed 2022-05-22 by LazeMSS