LazeMSS/OctoPrint-UICustomizer #250: Webcam display in right column is doubled.

Ompapamowmow, 2022-05-19:
[image]
## Description
After initial boot, when I change tabs, a second webcam view opens up right under the original in the right column. 
When I double click it, it opens the expanded pop-up view which is also doubled. When I double click again to get it to minimize it goes back to normal with only 1 webcam image. After that it works normally for quite some time, but double again every once in a while. I have not yet pinpointed a specific action or circumstance that causes it to double.
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.8.9**
- Terminal Messaging[terminalmessaging] v0.1.0
- Terminal Commands Extended[terminalcommandsextended] v0.1.6
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- Prusa Mini ETA Plugin[prusa_mini_eta] v0.1.4
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- PrettyGCode[prettygcode] v1.2.4
- Obico for OctoPrint[obico] v2.0.1
- FileManager[filemanager] v0.1.6
- Exclude Region[excluderegion] v0.3.0
- Active Filters Extended[active_filters_extended] v0.1.0
- Octolapse[octolapse] v0.4.1
- Consolidate Temp Control[consolidate_temp_control] v0.1.8
- Bed Visualizer[bedlevelvisualizer] v1.1.1

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.8.0
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/101.0.4951.64 Safari/537.36 Edg/101.0.1210.47

LazeMSS, 2022-05-20:
Could you try disabling the PrettyGCode plugin

LazeMSS, 2022-05-22:
Hopefully this is also fixed in the next update but please try disabling PrettyGCode and Octolapse to check if its something conflicting from them,

LazeMSS, 2022-05-25:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.0

Please reopen and comment if not working as expected.

MATV0, 2022-05-27:
Hi, I'm completely new to Github. Made this account to say that I've got exactly the same problem with the double webcams, and I'm using your 0.1.9.0. Sorry, I don't know how to reopen the above

LazeMSS, 2022-06-01:
> Hi, I'm completely new to Github. Made this account to say that I've got exactly the same problem with the double webcams, and I'm using your 0.1.9.0. Sorry, I don't know how to reopen the above

What other plugins do you have installed?

MATV0, 2022-06-16:
> > Hi, I'm completely new to Github. Made this account to say that I've got exactly the same problem with the double webcams, and I'm using your 0.1.9.0. Sorry, I don't know how to reopen the above
> 
> What other plugins do you have installed?

Hi, sorry new to all this. Is there a way I can export a list? Does this help:

[image]

Trying it just now, there was only one cam until I switched to the Pretty GCode tab and then it instantly doubled. Also I'm using UI Customiser, on a 3440x1440 screen.

MATV0, 2022-06-16:
Ah, how about this Exported list

[{"key": "uicustomizer", "name": "UI Customizer", "url": "https://github.com/LazeMSS/OctoPrint-UICustomizer"},
{"key": "touchui", "name": "TouchUI", "url": "https://github.com/BillyBlaze/OctoPrint-TouchUI"},
{"key": "terminalmessaging", "name": "Terminal Messaging", "url": "https://github.com/jeffeb3/OctoPrint-TerminalMessaging"},
{"key": "printscheduler", "name": "Print Scheduler", "url": "https://github.com/jneilliii/OctoPrint-PrintScheduler"},
{"key": "PrintJobHistory", "name": "PrintJobHistory", "url": "https://github.com/OllisGit/OctoPrint-PrintJobHistory"},
{"key": "preheat", "name": "Preheat Button", "url": "https://github.com/marian42/octoprint-preheat"},
{"key": "SpoolManager", "name": "SpoolManager Plugin", "url": "https://github.com/OllisGit/OctoPrint-SpoolManager"},
{"key": "SlicerSettingsParser", "name": "SlicerSettingsParser", "url": "https://github.com/tjjfvi/OctoPrint-SlicerSettingsParser"},
{"key": "PrintTimeGenius", "name": "PrintTimeGenius Plugin", "url": "https://github.com/eyal0/OctoPrint-PrintTimeGenius"},
{"key": "prettygcode", "name": "PrettyGCode", "url": "https://github.com/Kragrathea/OctoPrint-PrettyGCode"},
{"key": "GcodeEditor", "name": "GcodeEditor", "url": "https://github.com/ieatacid/OctoPrint-GcodeEditor"},
{"key": "extrafileinfo", "name": "ExtraFileInfo", "url": "https://github.com/larsjuhw/OctoPrint-ExtraFileInfo"},
{"key": "excluderegion", "name": "Exclude Region", "url": "https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin"},
{"key": "camerasettings", "name": "Camera Settings", "url": "https://github.com/The-EG/OctoPrint-CameraSettings"},
{"key": "bedcooldown", "name": "Bed Cooldown", "url": "https://github.com/rfinnie/OctoPrint-BedCooldown"},
{"key": "octoeverywhere", "name": "OctoEverywhere!", "url": "http://www.OctoEverywhere.com"},
{"key": "octoapp", "name": "OctoApp", "url": "https://github.com/crysxd/OctoApp-Plugin/"},
{"key": "marlingcodedocumentation", "name": "Marlin GCode Documentation", "url": "https://github.com/costas-basdekis/MarlinGcodeDocumentation"},
{"key": "DisplayLayerProgress", "name": "DisplayLayerProgress Plugin", "url": "https://github.com/OllisGit/OctoPrint-DisplayLayerProgress"},
{"key": "consolidate_temp_control", "name": "Consolidate Temp Control", "url": "https://github.com/jneilliii/OctoPrint-ConsolidateTempControl"},
{"key": "bedlevelvisualizer", "name": "Bed Visualizer", "url": "https://github.com/jneilliii/OctoPrint-BedLevelVisualizer"},
{"key": "IFTTT", "name": "OctoPrint-IFTTT", "url": "https://github.com/tjjfvi/OctoPrint-IFTTT"},
{"key": "cancelobject", "name": "Cancel Objects", "url": "https://github.com/paukstelis/OctoPrint-Cancelobject"}]

LazeMSS, 2022-06-16:
So it seems to be related to PrettyGCode  - I thought i had fixed that.

LazeMSS, 2022-06-18:
@MATV0  can you post screenshots? Does the problem only happen if you have PrettyGCode enabled? Can you recreate it again and again - if yes then how?

rbrian, 2023-01-29:
Mine is doing the same thing, on 1.19.0. Disabling PrettyGCode fixes it. Re-enabling, it comes right back, so yes seems to be consistent and related to PGC.

LazeMSS, 2023-02-03:
Basically pretty gcode seems to be dead and I can't fix it properly and OctoPrint is revamping the entire webcam stuff so I need to reimplement it all anyway.

LazeMSS, 2023-03-19:
This is expected to be fix in the upcoming 0.1.9.4 release combined with the OctoPrint 1.9.0rc3 and later versions.

LazeMSS, 2023-03-22:
This should now be fixed when 1.9.0 is released - closing for now - wait for 1.9.0 release or try the RC3 version.
https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.4

closed 2023-03-22 by LazeMSS