LazeMSS/OctoPrint-UICustomizer #252: Can't change Octoprint settings

tinker3433, 2022-05-28:
Hi, 
After upgrading OctoPrint to 18.1 I cannot properly change settings with UICustomizer enabled
When the dialog box opens i can change anything in the "Printer" section, including "General" "Intervals..." "Firmware" and "Behavior".  However, I cannot click the buttons to change to "Features" "Octoprint" or "Plugins".

If I boot OctoPrint in safe mode and disable UICustomizer the "tabs" move back to being a column on the left and everything works as expected.

OctoPrint 1.8.1
Python 3.7.3
OctoPi 0.18.0
UICustomizer 0.1.9.0

[image]

tinker3433, 2022-05-28:
other issues are also showing up. issue is not with UICustomizer. please delete issue

LazeMSS, 2022-05-28:
Okay

closed 2022-05-28 by LazeMSS