LazeMSS/OctoPrint-UICustomizer #253: TP-Link Smart Plug tab always shows up.

plduganaman, 2022-05-30:
## Description
**ENTER DESCRIPTION HERE
Describe your problem?
I have the TP-Link Smart Plug tab set to not show, however it does whenever I first go to Octoprint.
What is the problem?
I have the TP-Link Smart Plug tab set to not show, however it does whenever I first go to Octoprint
Can you recreate it?
Yes
Did you try disabling plugins?
No since that would remove the thing I am trying to test.
Did you remember to update the subject?**
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.9.0**
- TemperatureLegendMover[TemperatureLegendMover] v1.0.0
- TP-Link Smartplug[tplinksmartplug] v1.0.2
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- Navbar Temperature Plugin[navbartemp] v0.14
- ipOnConnect[ipOnConnect] v0.2.4
- Firmware Updater[firmwareupdater] v1.13.3
- DisplayZ[displayz] v0.2.0
- DisplayProgress[displayprogress] v0.1.3
- Cost Estimation[costestimation] v3.5.0

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.8.1
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/101.0.4951.67 Safari/537.36

plduganaman, 2022-05-31:
If I click on the settings (wrench icon) and then close the settings, the tab will disappear.  Maybe the order of operations affects this?   Maybe the TP_Link starts after the UI?

LazeMSS, 2022-05-31:
I will check

LazeMSS, 2022-06-01:
What settings are you using for the tp plugin - it works fine here but I don't have anything to setup the plugin.

jneilliii, 2022-06-18:
I use these plugins and cannot re-create your issue.

[image]

[image]

plduganaman, 2022-06-18:
Do you have a smart plug running?  In other words is the ip address of the plug active on the network?  BTW this is NOT any big deal to me just a minor annoyance.  I know it won't display on mine if it cannot locate the plug.

jneilliii, 2022-06-18:
Yes I do. It is a smart strip that controls multiple plugs and is reporting power data (therefore showing the tab). The plugin on it's own will not show the tab if there isn't any power data reported, because the tab is useless without that data.

plduganaman, 2022-06-18:
Well if I'm the only one with this issue I would close it.  It is super minor and no point in you trying to troubleshoot my system.

jneilliii, 2022-06-19:
try adding this to your advanced section in UI Customizer...

```
#tab_plugin_tplinksmartplug_link {
  display: none !important;
}

plduganaman, 2022-06-19:
That removed it alright. Thanks!

closed 2022-06-19 by plduganaman