LazeMSS/OctoPrint-UICustomizer #254: Cannot hide webcam in the Control Tab

linuxdudeg, 2022-06-07:
## Description
After updating UI Customizer to 0.1.9.0 I still have the same problem - the webcam is always displayed in the Control tab
I tried selecting and deselecting, but it's still visible.
At the moment I have the Control tab hidden so it's not a problem 
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.9.0**
- PrettyGCode[prettygcode] v1.2.4
- Firmware Updater[firmwareupdater] v1.13.3
- Dashboard[dashboard] v1.19.8
- Creality-2x-temperature-reporting-fix[ender3v2tempfix] v0.0.4
- Bed Visualizer[bedlevelvisualizer] v1.1.1

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.8.1
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/102.0.5005.61 Safari/537.36

linuxdudeg, 2022-06-09:
Hi Mikka,
I disables all plugins other than those from Gina but still the same problem. If I can help you in any way please let me know, maybe the source from the web page or look at specific elements on the page?
 Graham

LazeMSS, 2022-06-18:
@linuxdudeg can you try and do  a step by step explanation - ie:
is the webcam show on the control tab right after reloading or after you change tabs etc.?

linuxdudeg, 2022-06-19:
Hi Mikka,
No problem, here we go :
- Start OctoPi / Octoprint
- Connect from browser, my opening tab is the Dashboard, camera is visible
- Click on Control tab, camera is visible above the XYZ Controls
- UI Customizer.General Deselect Hide webcamera on Control tab, Save
- UI Customizer.General Select Hide webcamera on Control tab, Save
- webcam stream still visible, toggle between Dashboard and Control tab - still visible

I've tried many combinations of changing the settings, but the webcam stream is always visible in the Control Tab.

Could it be that the Dashboard webcam setting overrides the UI Customizer? Just a thought.

Hope this helps,

Graham

LazeMSS, 2022-06-19:
Well basically my plugin don't allow you to hide the main webcam unless you add the UI Customizer "widget" for showing the webcam as a panel. The idea when I made it and the latest changes was why would you wan't to have a webcam configured and not shown anywhere - but you wan't the webcam shown on the dashboard and nowhere else?

It works fine with disabling when i run dashboard and have main tabs like this:
[image]

LazeMSS, 2022-12-29:
@linuxdudeg - sorry for the long delay - do you also have this problem when Dashboard is disabled?

7s3, 2023-03-12:
I'm having the same problem. Webcam either has to be at the main dashboard/control tab, or in one of the columns.
No way to completely hide it...

LazeMSS, 2023-03-13:
> I'm having the same problem. Webcam either has to be at the main dashboard/control tab, or in one of the columns. No way to completely hide it...

To completely hide the webcam try disabling the webcam in the webcam settings?

7s3, 2023-03-13:
> To completely hide the webcam try disabling the webcam in the webcam settings?

"Enable webcam support" in #settings_webcam ?
Possibly fixed, but I remember already unchecking that and somehow it became enabled again.
I will write back if it reverts and starts having the issue again.

LazeMSS, 2023-03-19:
This is expected to be fix in the upcoming 0.1.9.4 release combined with the OctoPrint 1.9.0rc3 and later versions.

LazeMSS, 2023-03-22:
This should now be fixed when 1.9.0 is released - closing for now - wait for 1.9.0 release or try the RC3 version.
https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.4

closed 2023-03-22 by LazeMSS