LazeMSS/OctoPrint-UICustomizer #255: Icons for other plugin tabs are not showing despite being set.

jackmoxley, 2022-06-08:
## Description
***Describe your problem?*** Icons for other plugin tabs are not showing despite being set.
***What is the problem?*** I have Pretty Gcode and Bed visualizer installed, I have set icons up for them, without colour, and the do not appear on the screen despite a refresh.  See images at the bottom, note GCode 3D and Bed are without icons. 
Love the tool btw.
***Can you recreate it?*** Yes
***Did you try disabling plugins?*** No
***Did you remember to update the subject?*** Good point
<hr>

**Plugins installed**
- ABL Expert Plugin[ABL_Expert] v0.6
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- BLTouch Plugin[BLTouch] v0.3.4
- Creality-2x-temperature-reporting-fix[ender3v2tempfix] v0.0.4
- Fullscreen Plugin[fullscreen] v0.0.6
- Preheat Button[preheat] v0.8.0
- PrettyGCode[prettygcode] v1.2.4
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- Themeify[themeify] v1.2.2
- Top Temp[toptemp] v0.0.1.7
- **UI Customizer[uicustomizer] v0.1.9.0**
- Z Probe Offset[z_probe_offset] v0.5

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: true

**Software versions**
- OctoPrint 1.8.1
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/102.0.0.0 Safari/537.36

[image]
[image]

LazeMSS, 2022-06-09:
Notice the dropdown on the right side - it says text-only

jackmoxley, 2022-06-09:
Completely missed that, sorry. Will close