LazeMSS/OctoPrint-UICustomizer #256: Hide Webcam on Control tab

BH471, 2022-06-11:
When 'Hide Webcam on Control tab' is checked it is gone when previewing but returns once saved or Octoprint is rebooted/restarted. 

**Plugins installed**
- TouchUI[touchui] v0.3.18
- Gcode Macros[gcode_macro] v1.1.1
- Active Filters Extended[active_filters_extended] v0.1.0

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.8.1
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/102.0.5005.63 Safari/537.36 Edg/102.0.1245.33

LazeMSS, 2022-06-17:
Can you explain a bit more - is it never shown in the sidebar or don't you have the sidebar webcam on?

LazeMSS, 2022-06-18:
Duplicate/similar to: #254

closed 2022-06-18 by LazeMSS

LazeMSS, 2022-06-18:
@BH471 just want to make sure - do you have the webcam disabled in main control and also not enabled the webcam widget? Why not complete disable the webcam then?

BH471, 2022-06-19:
I think I have the webcam turned off in all possible settings but still have the blank webcam preview in the control tab. That being I have un-checked 'Enable Webcam, support' in the 'Webcam and Timelapse' feature and I have it disabled in the UI customizer settings for Timelapse also.