LazeMSS/OctoPrint-UICustomizer #257: USB cam[BUG]

wjflier, 2022-06-20:
**PLEASE USE THE "Report a bug" button found in settings**

That button will prefill everything and you don't have to do anything besides describe the problem.

**Describe the bug**
USB cam doesn't load in side panel

**To Reproduce**
Steps to reproduce the behavior:
1. What settings are you using in UICustomizer
2. Go to '...'
3. Click on '....'
4. Scroll down to '....'
5. See error

**Expected behavior**
webcam to load

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] win10
 - Browser [e.g. chrome, safari]opera
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
using raspberry 4 with usb camera, shows up in control, but switching to pretty gcode, the cam doesn't load in Pretty gcode or, on the side panel.

LazeMSS, 2022-06-22:
Does it work without UI Customizer

wjflier, 2022-06-23:
Got it

LazeMSS, 2022-07-13:
@wjflier can we close this?

captainwasabi, 2022-08-23:
I am having this problem. It was working, then it stopped.  The camera shows up on the dashboard and in the controls panel, but not the side panel.

closed 2022-12-29 by LazeMSS