LazeMSS/OctoPrint-UICustomizer #258: UI Customizer Settings Crashes upon saving when attempting to put all controls in the Left-Width=0, Center-Width=12, Right=0

badonj001, 2022-06-30:
**Description:** UI Customizer Settings Crashes upon saving when attempting to put all controls in the Left-Width=0, Center-Width=12, Right=0

**To Reproduce**
0.) **Go to:** 'Settings> Plugins> UI Customizer> [Layout]'
1.) Move all controls to the center set the center width to '12'
2.) If you can change the left and right columns to '0'
3.) Click Save and you hopefully will see the error in the screenshot below.
4.) I only say hopefully because if it does it to you, you can hopefully back track to the reason for the total width of '12' Rule.
5.) It would be awesome to have all controls in center for scrolling.

**Expected behavior**
Under Layout I was attempting to add all of the controls to the center attempting to follow the rule the total can only be 12. If I can get all them to the center and adjust the center to 12, then the error in the below screenshot is what happens and then OctoPrint Settings windows disappeared not applying the changes.

**Screenshots:**
[image]
[image]

**Plugins installed**
- Macro[macro] v0.3.2
- **UI Customizer[uicustomizer] v0.1.9.0**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: false
- fluidLayout: true
- gcodeFullWidth: true
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.8.1
- Python 3.8.10

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.0.0 Safari/537.36

badonj001, 2022-06-30:
Submitting the bug helped me give it some more thought and something I noticed when I first start using UI Customizer and adjusting scaling with the UI Layout. 

1.) So what I just did was put all of the controls in the Left Column, and had to leave the left with  with a width '11',  and leave the center with a width of '1' as it would not let me set it to '0'.

2.) I saved and returned to the UI.

3.) All of the controls were on the left and there was the padded space of the Center.

4.) I increased the Page Scaling until the padded column on the right went away and gave the bonus of larger buttons to press.

5.) I then went onto UI Customizer's Layout settings and the Center and Right Columns were set to '0',, and the Left Width was set to '11' still.

6.) I changed the Left Width to '11' and save it without crashing.

7.) I was then able to scale down the page to a comfortable setting for me. and now I have one column after a bit of 'turn this, wiggle that, in this order now do this step again.

So I guess if it can be done with settings being changed in a certain order in UI Customizer and Mozilla (Firefox, Chrome, Explorer Edge), then maybe easier for you to track down why it works like that, then maybe you can fix it?

I am not the best at it, but I will try to make a video showing you the different approaches I took and at what point the outcome changed. It was a bit of a journey to figure that out.

**Screenshots:**
[image]
[image]
[image]

-/retnel

LazeMSS, 2022-07-01:
I will look ASAP.

LazeMSS, 2022-07-04:
Seems like you found one of the bugs i have been chasing for a long time - will try and fix later today

LazeMSS, 2023-03-05:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.1 - please reopen if not working.

closed 2023-03-05 by LazeMSS