LazeMSS/OctoPrint-UICustomizer #259: Move main tab items to left or right column

gee4vee, 2022-07-25:
I'm not sure if this is possible, but it would be nice if you could place some of the things that are in the main tabs group in either the left or right column. For example, I use the Pretty gcode viewer and I would like it to be on the left or right so it can be shown with the webcam at the same time. Is this possible with current settings? I'm not sure what determines what goes in the main center tabs. Thanks for a great plugin!

kerl4real, 2023-08-31:
I agree, I was looking for this same option specifically for PrettyGCode and Dashboard. 
It would be ideal if any widget that can be toggled on/off in the "Main Tabs" can also be toggled on/off in the "Layout". 
I've tried looking at the code but it would take me awhile just to understand how everything's being built.

csm10495, 2024-09-16:
This would be a great feature. I'd love to be able to move PrettyGCode to the side.. even having something like terminal on the side would be slick.

LazeMSS, 2024-09-17:
Its not that easy to do. I will look into it later this month.

Pipow-fmf, 2025-08-22:
Queria saber se já foi resolvido isso?