LazeMSS/OctoPrint-UICustomizer #261: Temperature presets outside screen area

phptanker, 2022-08-15:
## Description
Describe your problem?
What is the problem?
Can you recreate it?
Did you try disabling plugins?
Did you remember to update the subject?**

When having more than 3 presets for temperatures in the control section, it is impossible to see more than 3 of the presets. See sreenshot. 
[image]

<hr>

**Plugins installed**
- Backup Scheduler[backupscheduler] v0.1.0
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Consolidate Temp Control[consolidate_temp_control] v0.1.9
- Firmware Updater[firmwareupdater] v1.13.3
- Obico for OctoPrint[obico] v2.1.2
- Octolapse[octolapse] v0.4.1
- Print Scheduler[printscheduler] v0.0.4
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- RGB Status[rgb_status] v1.7.2
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- Terminal Commands Extended[terminalcommandsextended] v0.1.7
- **UI Customizer[uicustomizer] v0.1.9.0**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.8.2
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/104.0.5112.83 Safari/537.36

phptanker, 2022-08-16:
Works when changing webcam to never display in control area.

phptanker, 2022-08-18:
MY mistake: This still is an issue, I was just lucky with my scroll bar covering more area than the box. Still doesn't display the whole drop-down menu.

LazeMSS, 2022-12-29:
I can't recreate this no matter what I try.
- what theme are you using?
- is this happening only inside the consolidate control tab window?

My guess is that some other plugin is changing the css - could you try disabling one after another?

phptanker, 2022-12-29:
Hi @LazeMSS, using Discorded theme. Exactly, the next troubleshooting steps for me then is:
1. Disable the consolidate temp/control plugin and see if the issue persists. 
2. Disable more plugins until the issue disappears. 

I'll try to get back to you asap.

LazeMSS, 2022-12-30:
Awesome and sorry for slow response

phptanker, 2023-01-04:
The issue seems to persist with the regular temperature view in the discorded theme. But honestly this feels like more of a visual glitch than a problem. The workaround is to expand any of the tabs to the left/right and I'm able to see the whole list of temperatures. 

@LazeMSS in your opinion, do you think it's even worth keeping this issue opened? It's very isolated and I dont mind closing.

LazeMSS, 2023-01-04:
I will test a bit more but closing for now ;)

closed 2023-01-04 by LazeMSS