LazeMSS/OctoPrint-UICustomizer #264: Custom Color in Theme

tcon2010, 2022-09-27:
I'm using the discoranged theme and simply wanted to change the orange to a new color. I looked at the inspector and saw that it was using --accent from the root of the css. Knowing that, I went to advanced and entered 
`:root {--accent: #aaf0d1 !important;}`
but that did not work upon saving and refreshing. I went back to advanced and just to see what would happen I checked preview and _it worked_. I clicked save and all was good... until I refreshed the page. It went right back to the orange. Is there a simple way of changing colors in the theme as this doesn't seem to be working?

[image]
[image]

LazeMSS, 2022-09-27:
Its basically because the order of the CSS is handled - I will look into a workaround but Im currently on a break. But the planned feature is to allow customizing the colors in the themes instead of the CSS overwrites

j-snijder, 2022-11-17:
Hi, maybe a tip for use in the meanwhile.

Although octoprint warns you for conflicting plugins you could reinstall Themify ( I did :) ) and use thast together with UIcustomizer.

In my case it worked and I got back the things I altered in Themify (which I was using before I discovered UIcustomizer was preferred bij the Dev.

stayingarmed, 2022-12-11:
Did anything more come of this? I would really like to customize theme colors!

LazeMSS, 2022-12-12:
I just did a full rework of the bugs and features list on another plugin. This week I will go through the UI customizer backlog

LazeMSS, 2022-12-28:
I just started working on the bug backlog - this appears to be as stated by @tcon2010 by using important - even after reloading.
I'm still planning on making an easier customizing where one can specify the color variables

LazeMSS, 2022-12-29:
Closing this as it is working as intended - I will improve it better when "color variables" are customizable

LazeMSS, 2022-12-29:
https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/217

closed 2022-12-29 by LazeMSS