LazeMSS/OctoPrint-UICustomizer #265: UI Customizer has issues keeping the camera stream loaded

Tech0verlord, 2022-10-10:
## Description
**ENTER DESCRIPTION HERE
Describe your problem?
Webcam stream randomly unloads.

What is the problem?

So after updating UI customizer from the last time I updated in July, I get an issue with the webcam stream, most of the time, when loading into Octoprint, the stream won't load automatically and I have to manually navigate to the stream, at which point it shows up in the Octoprint UI. Always says it's a configuration error even though the URL is correct, as clicking the linked URL in the webcam window brings up the webcam.  It always happens when first loading up Octoprint's UI, and sometimes also randomly after the webcam stream is up and the window hasn't been changed or navigated away from. Other times the webcam stream will be loaded, but when Octoprint's UI finishes loading, the stream unloads and asks if the configuration is correct until I do something to forcibly refresh the stream (change Octoprint tabs, browser tabs, etc.)

Can you recreate it?
Yes

Did you try disabling plugins?
No

Did you remember to update the subject?**
Yes, recent updates have caused the issues.
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.9.0**
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- Cura Thumbnails[UltimakerFormatPackage] v1.0.2
- PSU Control[psucontrol] v1.0.6
- PSU Control - TPLink[psucontrol_tplink] v1.0.1
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- FileManager[filemanager] v0.1.6
- M73 Progress Plugin[m73progress] v0.2.1
- Bed Visualizer[bedlevelvisualizer] v1.1.1

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: false
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: true
- themeLocal: false

**Software versions**
- OctoPrint 1.8.4
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/105.0.0.0 Safari/537.36
[image]
[image]

Tech0verlord, 2022-10-10:
Upon messing around a bit. I noticed the stream won't unload as long as the webcam widget on the right is expanded, even if nothing is displaying because the control tab (my default tab) is open. If I shrink the widget, the stream closes again shortly afterwards.
 
[image]

Tech0verlord, 2022-10-16:
I would just like to note that if I roll back to 1.8.9 the problem goes away.

LazeMSS, 2022-10-17:
Will look asap - im very busy with my work at the moment

Tech0verlord, 2022-10-17:
No worries, just noting what I'm finding. Gonna hang back on 1.8.9 for the time being. I'm assuming the bug might be related to fix for #246.

Revenger, 2022-11-13:
Having a similar issue.
Webcam works fine on dash but in sidebar and control doesn't with ui customiser.

[image]
[image]

Revenger, 2022-11-14:
Another issue is this.

[image]

Tech0verlord, 2022-11-14:
Try installing 1.8.9, see if that fixes it.

LazeMSS, 2022-11-14:
I'm very busy at the moment - I will look ASAP

j-snijder, 2022-11-16:
Same here, where can I find v 1.8.9 ?

j-snijder, 2022-12-05:
> 

Maybe, if it's okay with you, you could report on github you couold need some help on this ?

Just my 2 cents :)

LazeMSS, 2022-12-05:
Should i report it or?

j-snijder, 2022-12-05:
I would advise you to do yes. Its your plugin 😊

Kind regards.

Verzonden vanuit Mail voor Windows

Van: LazeMSS
Verzonden: maandag 5 december 2022 12:40
Aan: LazeMSS/OctoPrint-UICustomizer
CC: j-snijder; Comment
Onderwerp: Re: [LazeMSS/OctoPrint-UICustomizer] UI Customizer has issues keepingthe camera stream loaded (Issue #265)

Should i report it or?
—

Dit e-mailbericht is met Avast-antivirussoftware gecontroleerd op virussen.
www.avast.com

LazeMSS, 2022-12-06:
Should i report on github i need som help with code - its right here on github - anycone can clone and do fixes

j-snijder, 2022-12-06:
Hi 😁Out of respect most, if not all, don't touch your code I guess.Asking for help because you're too busy at the moment could help you and the users of this magnificent code/plugin.Just a thought though 😁

LazeMSS, 2022-12-29:
@j-snijder what plugins do you have installed?

@Revenger  could you try and disable the dashboard plugin - it does alot of hacks with the webcam feed.

@Tech0verlord  I really can't recreate on my dev and live machines - either is because something is really slow or something else - tried installing the same plugins on a clean install and using the same browser.

LazeMSS, 2023-03-19:
This is expected to be fix in the upcoming 0.1.9.4 release combined with the OctoPrint 1.9.0rc3 and later versions.

LazeMSS, 2023-03-22:
This should now be fixed when 1.9.0 is released - closing for now - wait for 1.9.0 release or try the RC3 version.
https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.4

closed 2023-03-22 by LazeMSS