LazeMSS/OctoPrint-UICustomizer #268: [FR] Cancel Button Actions

SemperFi89, 2022-11-27:
I would like the ability to configure the actions that are taken when the user hits the cancel button in the State box.  Currently when you hit cancel it just stops right where it is and shuts off the heaters.  For me, I want it to move X to **0** and Y to (either **max** or a fixed number if you can't use a variable).

LazeMSS, 2022-11-28:
Maybe take a look here:
https://plugins.octoprint.org/plugins/inlineconfirm/

SemperFi89, 2022-11-29:
> Maybe take a look here: https://plugins.octoprint.org/plugins/inlineconfirm/

Per the description, it only will get rid of the confirmation box of the cancel.  I want to be able to send something like G0 X 0 Y 235 once the cancellation is confirmed to put the print head at the far left and move the bed completely out.  I don't care if there is a popup to confirm or if I have to double-click cancel.

LazeMSS, 2022-12-03:
Sorry but is this not handled in OctoPrint Settings > Printer > GCODE Scripts > After print job is cancelled?

SemperFi89, 2022-12-03:
It is, i found that later.  thanks for your help

Sent from Nine<http://www.9folders.com/>

closed 2022-12-03 by LazeMSS