LazeMSS/OctoPrint-UICustomizer #269: HLS Wyzecam Stream not showing up on dashboard but works within settings.

msrlrocket, 2022-11-29:
Hi, I have an HLS steam that I am trying to use for the webcam. This camera works when I have UICustomizer disabled and does not when its enabled. Upon opening Octoprint, It will show the controls for the printer jammed at the top with two little volume and expand buttons but no stream. I have seen that adding a new camera plugin that shows on the right-hand side will also show the camera.

msrlrocket, 2022-11-29:
Discovered that if I click the small PIP button at the top since it looks like the CSS collapsed it will open in a small window with camera working.

LazeMSS, 2022-11-30:
I will try and test ASAP

LazeMSS, 2023-03-19:
This is expected to be fix in the upcoming 0.1.9.4 release combined with the OctoPrint 1.9.0rc3 and later versions.

LazeMSS, 2023-03-22:
This should now be fixed when 1.9.0 is released - closing for now - wait for 1.9.0 release or try the RC3 version.
https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.4

closed 2023-03-22 by LazeMSS