LazeMSS/OctoPrint-UICustomizer #270: Memory Usage

muckypaws, 2022-12-17:
## Description
macOS Safari complains too many resources used when UI Customiser is open for a period of time.

macOS Safari slows down when UI Customiser is open for a period of time > 30minutes

Can you recreate it? Every time UI Customiser plug-in is enabled.
Did you try disabling plugins? Yes, disabling UI Customiser prevents the issue.

Hope So
<hr>

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Calibration Tools[CalibrationTools] v1.0.4
- Navbar Clock[navbarclock] v1.1.4
- Navbar Temperature Plugin[navbartemp] v0.15
- TouchUI[touchui] v0.3.18
- **UI Customizer[uicustomizer] v0.1.9.0**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: true

**Software versions**
- OctoPrint 1.8.6
- Python 3.9.2

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.6.1 Safari/605.1.15

LazeMSS, 2022-12-27:
Yes it due to the way mjpeg causes mem leaks in Safari - I will look ASAP

LazeMSS, 2022-12-29:
Can you try and look with the inspector open - my own Safari is not complaining but will try on another machine

LazeMSS, 2023-03-19:
This is expected to be fix in the upcoming 0.1.9.4 release combined with the OctoPrint 1.9.0rc3 and later versions.

LazeMSS, 2023-03-22:
This should now be fixed when 1.9.0 is released - closing for now - wait for 1.9.0 release or try the RC3 version.
https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.4

closed 2023-03-22 by LazeMSS