LazeMSS/OctoPrint-UICustomizer #271: [BUG] Prusa tribute theme - weirdness with Terminal autoscroll

synman, 2022-12-18:
## Description
It is like autoscroll is out to drive me crazy.  It took this happening 5 or 6 times for me to stop second guessing myself about hitting it accidentally.  It mysteriously auto disables autoscroll whenever anything new hits the terminal window after toggling autoscroll for the first time.  From that point forward, it is bugged out.

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.9.0**
- Timelapse Purger[timelapsepurger] v0.1.4
- Terminal Commands Extended[terminalcommandsextended] v0.1.7
- Signalclirestapi Plugin[signalclirestapi] v0.1.1
- System Command Editor[systemcommandeditor] v0.3.4
- Resource Monitor[resource_monitor] v0.3.9
- PSU Control[psucontrol] v1.0.6
- GcodeEditor[GcodeEditor] v0.2.12
- FileManager[filemanager] v0.1.6
- Custom Control Editor[customControl] v0.2.2
- CNC GCode Viewer[cnc_gcodeviewer] v1.1.1
- Better Grbl Support[bettergrblsupport] v2.3.0rc1

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: false
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: true
- themeLocal: false

**Software versions**
- OctoPrint 1.8.6
- Python 3.9.2

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/108.0.0.0 Safari/537.36

LazeMSS, 2022-12-28:
I can't recreate - does this also happen with other themes? 
Could you step by step explain what you do or not do - ie. you click autoscroll and then as soon data is sent to terminal autoscroll is disabled?

synman, 2022-12-28:
Nope, just this one.  I'll see if it's conflicting with something else. 

If I come up empty, I'll record a short video.

LazeMSS, 2022-12-28:
Thanks. I might have seen it once now. Im not sure but it was triggered when I clicked a file to print

LazeMSS, 2023-03-05:
Can't recreated and closing - @synman  reopen it you see it again or have more info.

closed 2023-03-05 by LazeMSS