LazeMSS/OctoPrint-UICustomizer #274: Can't get 3 columns

ackerthehacker2, 2023-01-29:
## Description
It was all working ..did an upgrade ...and now the most I can get is 2 columns ..the third one is below ... I tried deinstall and reinstalling.
<hr>

**Plugins installed**
- Action Commands[actioncommands] v0.5
- Autoscroll[autoscroll] v0.0.3
- Backup Scheduler[backupscheduler] v0.1.0
- Bed Ready[bedready] v0.1.5
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Cancel Objects[cancelobject] v0.4.7
- Change Filament Plugin[Change_Filament] v0.3.2
- Consolidate Temp Control[consolidate_temp_control] v0.1.9
- Consolidated Tabs[consolidatedtabs] v1.0.1
- Cost Estimation[costestimation] v3.5.0
- Custom Control Editor[customControl] v0.2.2
- Custom GCode Events[custom_gcode_events] v0.3.0
- Dashboard[dashboard] v1.19.9
- Detailed Progress[detailedprogress] v0.2.7
- Detector2 Plugin[detector2] v0.1.4
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- DisplayZ[displayz] v0.2.0
- Draggable Files[draggable_files] v1.1.2
- Dragon Order[dragon_order] v0.1.7
- Exclude Region[excluderegion] v0.3.2
- ExtraFileInfo[extrafileinfo] v1.3.0
- FileManager[filemanager] v0.1.6
- Firmware Updater[firmwareupdater] v1.13.3
- Gcode Macros[gcode_macro] v1.3.0
- Gcodebar Plugin[gcodebar] v0.1.5
- GcodeEditor[GcodeEditor] v0.2.13
- Google Drive Backup[googledrivebackup] v0.1.1
- Google Drive Files[googledrivefiles] v0.1.0
- HomeAssistant Discovery[homeassistant] v3.6.2
- ipOnConnect[ipOnConnect] v0.2.4
- M73 Progress Plugin[m73progress] v0.2.1
- Macro[macro] v0.3.2
- Macro Steps[MacroSteps] v0.1.1
- Marlin EEPROM Editor[eeprom_marlin] v3.3.0
- Marlin GCode Documentation[marlingcodedocumentation] v0.18.0
- MQTT[mqtt] v0.8.13
- Navbar Clock[navbarclock] v1.1.4
- Navbar Temperature Plugin[navbartemp] v0.15
- Network Health Monitor[networkhealth] v1.0.4
- Obico for OctoPrint[obico] v2.3.0
- OctoDash Companion[octodashcompanion] v0.0.7
- OctoPod Plugin[octopod] v0.3.14
- Octoprint Display ETA[display_eta] v2.2.0
- OctoPrint-IFTTT[IFTTT] v1.4.1
- OneDrive Backup[onedrive_backup] v1.0.1
- OneDrive File Sync[onedrive_files] v1.0.5
- Persist Terminal Command History Plugin[persist_terminal_command_history] v0.2.0
- Power Failure Recovery[powerfailure] v1.1.0
- Preheat Button[preheat] v0.8.0
- PreHeat Helper[preheathelper] v0.1.2
- PrettyGCode[prettygcode] v1.2.4
- Print History Plugin[printhistory] v1.3
- PrintJobHistory[PrintJobHistory] v1.17.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.3.0
- PrintWatch[printwatch] v1.2.11
- Progress Title[progress_title] v0.1.2
- Prusa Leveling Guide[PrusaLevelingGuide] v1.0.18
- Resource Monitor[resource_monitor] v0.3.11
- Shutdown Printer[shutdownprinter] v1.0.9
- Sidebar Webcam[webcamSB] v0.1.8
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- Slicer Estimator[SlicerEstimator] v1.5.9
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.1
- SlicerSettingsParser[SlicerSettingsParser] v3.0.1
- Smart Preheat[smartpreheat] v0.0.6
- SpoolManager Plugin[SpoolManager] v1.7.0
- STL Viewer[stlviewer] v0.4.2
- System Command Editor[systemcommandeditor] v1.0.0
- TemperatureLegendMover[TemperatureLegendMover] v1.0.0
- Tempsgraph Plugin[tempsgraph] v0.3.7
- Terminal Commands Extended[terminalcommandsextended] v0.1.7
- Terminal Messaging[terminalmessaging] v0.1.0
- Thingiverse Downloader[thingiverse_downloader] v0.3.0
- Thingiverse Plugin[thingiverse] v1.1.0
- Timelapse Purger[timelapsepurger] v0.1.4
- TimeToFilament Plugin[TimeToFilament] v1.2.1
- Top Temp[toptemp] v0.0.2.3
- **UI Customizer[uicustomizer] v0.1.9.0**
- WiFi Status[wifistatus] v1.7.2

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: true
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: false
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.8.6
- Python 3.9.2
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/109.0.0.0 Safari/537.36

ackerthehacker2, 2023-01-29:
The screen is for the total of the columns to be less than 12 (i.e 11) ...  12 doesn't work, 11 does.

LazeMSS, 2023-01-29:
Try disabling sidebar webcam plugin. 
Or clear the settings for my plugins. 

Do you a warning message when saving?

I'm on Holliday so can't test anything.

ackerthehacker2, 2023-01-29:
I ditched a few plugins and it seemed to work again .....  Printwatch did an upgrade a few days ago so I am suspecting that.

LazeMSS, 2023-01-30:
You also have really many plugins... :)

closed 2023-02-03 by LazeMSS