LazeMSS/OctoPrint-UICustomizer #276: Blue Moon Theme UI Customizer

dut97, 2023-02-06:
## Description
**ENTER DESCRIPTION HERE
Describe your problem? File doesn't turn green after print is finished
What is the problem? Pure cosmetic
Can you recreate it? Yes
Did you try disabling plugins? No need
Did you remember to update the subject? Yes :)
<hr>

**Plugins installed**
- Autoscroll[autoscroll] v0.0.3
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- BLTouch Plugin[BLTouch] v0.3.4
- Creality-2x-temperature-reporting-fix[ender3v2tempfix] v0.0.4
- Dashboard[dashboard] v1.19.9
- M73 Progress Plugin[m73progress] v0.2.1
- Navbar Temperature Plugin[navbartemp] v0.15
- Network Health Monitor[networkhealth] v1.0.4
- OctoEverywhere![octoeverywhere] v1.11.1
- Octolapse[octolapse] v0.4.1
- Octoprint Display ETA[display_eta] v2.2.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.3.0
- PSU Control[psucontrol] v1.0.6
- PSU Control - RPi.GPIO[psucontrol_rpigpio] v1.0.4
- Resource Monitor[resource_monitor] v0.3.11
- **UI Customizer[uicustomizer] v0.1.9.0**
- WebcamTab[webcamtab] v0.3.0
- WiFi Status[wifistatus] v1.7.2
- Z Probe Offset[z_probe_offset] v0.5

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: false
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.8.6
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/109.0.0.0 Safari/537.36

LazeMSS, 2023-03-05:
It does on my setup - could you post a screenshot?

LazeMSS, 2023-07-25:
Closing - feedback missing

closed 2023-07-25 by LazeMSS