LazeMSS/OctoPrint-UICustomizer #280: Can't change name+icon on 3rd party plugins

Ariosinm, 2023-03-05:
## Description
I've been trying to change the name and also name+icon feature on 3rd party plugins and I can't do it in anyway I've tried. I'm pretty new to octopi and also to this pluggin. Sorry if I'm forgeting anything obvious but I can't find a way to do it.
**
<hr>

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Calibration Tools[CalibrationTools] v1.0.4
- Cancel Objects[cancelobject] v0.4.7
- Change Filament Plugin[Change_Filament] v0.3.2
- Cost Estimation[costestimation] v3.5.0
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- Enclosure Plugin[enclosure] v4.13.2
- FileManager[filemanager] v0.1.6
- Firmware Updater[firmwareupdater] v1.13.3
- GPIO Control[gpiocontrol] v1.0.8
- GPIO RGB Controller[gpiorgbcontroller] v0.3.3
- M300 Player[M300Player] v0.1.1
- OctoApp[octoapp] v1.1.0
- OctoEverywhere![octoeverywhere] v2.3.7
- Octolapse[octolapse] v0.4.1
- OctoRelay[octorelay] v1.2.0
- Power Failure Recovery[powerfailure] v1.1.1
- PrettyGCode[prettygcode] v1.2.4
- Print Scheduler[printscheduler] v1.0.1
- Printer Stats[stats] v3.1.3
- PrintJobHistory[PrintJobHistory] v1.17.0
- PSU Control[psucontrol] v1.0.6
- Resource Monitor[resource_monitor] v0.3.12
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.1
- SpoolManager Plugin[SpoolManager] v1.7.0
- Telegram Notifications[telegram] v1.6.5
- TP-Link Smartplug[tplinksmartplug] v1.0.2
- **UI Customizer[uicustomizer] v0.1.9.1**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: true
- themeLocal: false

**Software versions**
- OctoPrint 1.8.7
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/110.0.0.0 Safari/537.36

LazeMSS, 2023-03-06:
Which icons? Screenshot maybe ;)

Ariosinm, 2023-03-12:
[image]
I was looking to change all the "text only" gaps into "icon+text" but I can't, the other options on the list shows unmarked and it doesn't let you change it. And when I change the name of the spools to "bobinas" it isn't showing up in the main view

LazeMSS, 2023-03-12:
You need to pick and icons using the magnify glass before you can pick the other options that require an icon.

The label of the spools should be possible - maybe something in that plugin updates it ie. does it work for other tabs?

LazeMSS, 2023-03-19:
@Ariosinm ?

Ariosinm, 2023-04-04:
I've managed to change it for other tabs, but spools didn't change. Sorry
for the delay :(

LazeMSS, 2023-07-25:
Yes if it this plugin https://plugins.octoprint.org/plugins/SpoolManager/ then it automatically updates the icons if i'm not mistaken.

closed 2023-07-25 by LazeMSS