LazeMSS/OctoPrint-UICustomizer #281: [BUG] Compatibility issue with OctoPrint 1.9 - webcam not working

giuliolibrando, 2023-03-08:
**Describe the bug**

OctoPrint 1.9rc2  has moved Webcam integration to a plugin interface.
[https://octoprint.org/blog/2023/03/07/new-release-candidate-1.9.0rc1/](Release info)

Webcam View is not working, keep showing "Webcam stream loading..".
[image]

It works without the UI Customizer plugin.

NewsGuyTor, 2023-03-08:
Please don't remove the webcam functionality, it is vital for so many people! If the worst come to worst, and it is s too much work, please implement the simple workaround I detail here:

1. [image]

2. [image]

3. [image]

Replacing the HTML with the image HTML is a great workaround at least, a way better option that removing the webcam functionality.

LazeMSS, 2023-03-09:
@NewsGuyTor its not that easy but I have already fixed it in dev I hope

CornFlakes1991, 2023-03-09:
Even after updating to this fix, this issue persists

jneilliii, 2023-03-09:
@CornFlakes1991 is it the same error relative to "subscribe" as in the original error post in your browser's developer tools console tab?

jneilliii, 2023-03-09:
I suspect this may be a race condition or something. I just checked and this is the error I see in console tab with the latest version. However, if I run the command that is being assigned to OrgTabChange it is a function. 

[image]

CornFlakes1991, 2023-03-10:
> 

[image]

LazeMSS, 2023-03-10:
@CornFlakes1991 on RC2 or normal?

jneilliii, 2023-03-10:
Looks like 1.9.0 based on friendship warning. Error is prettygcode, disable that plugin and check again.

CornFlakes1991, 2023-03-10:
> @CornFlakes1991 on RC2 or normal?

RC2

> Looks like 1.9.0 based on friendship warning. Error is prettygcode, disable that plugin and check again.

Disabled it and still not webcam image :/

jneilliii, 2023-03-10:
Yes, this was what I was seeing too with the patched files. There's still something missing relative to keeping the stream live when not on the control tab. I'll be looking at this over the weekend for all my impacted plugins and will have a better idea of how the new classic webcam plugin is controlling this.

jneilliii, 2023-03-10:
maybe we could ask @crysxd what is needed to keep the stream active...

crysxd, 2023-03-10:
The problem is that you can now have more than one webcam and OctoPrint has a drop down to select the shown webcam. The controls VM will stop the webcam stream of inactive webcams, similarly to when you move away from the OctoPrint tab in your browser. The corresponding code is here:

- https://github.com/OctoPrint/OctoPrint/blob/95e87eb1970d8e322738f9698c9d665d78f2d74e/src/octoprint/static/js/app/viewmodels/control.js#L58
- https://github.com/OctoPrint/OctoPrint/blob/95e87eb1970d8e322738f9698c9d665d78f2d74e/src/octoprint/static/js/app/viewmodels/control.js#L84
- https://github.com/OctoPrint/OctoPrint/blob/95e87eb1970d8e322738f9698c9d665d78f2d74e/src/octoprint/static/js/app/viewmodels/control.js#L109

fly74, 2023-03-16:
The other thing is that the option "Hide webcamera on Control tab" don't hide it in OP 1.9.0rc2.

LazeMSS, 2023-03-17:
@fly74 did you turn on the webcam widget?

fly74, 2023-03-17:
@LazeMSS What do mean?

It is emty in the Main view:

[image]

And shown in the control section, even if hide option is set:

[image]

Enabled here?:

[image]

LazeMSS, 2023-03-17:
@fly74  on the RC2 or normal 1.8.2 octoprint?

fly74, 2023-03-18:
> @fly74  on the RC2 or normal 1.8.2 octoprint?
> 
> 

In the rc2. When I rollback to 1.8.2 everything is fine.

LazeMSS, 2023-03-18:
Ohh i might have that fixed in the upcoming release but that requires the rc3 to be released first

closed 2023-03-22 by LazeMSS

fly74, 2023-03-22:
@LazeMSS I can confirm, it works now. thx.

abulka, 2023-04-05:
> This should now be fixed when 1.9.0 is released - closing for now - wait for 1.9.0 release or try the RC3 version. https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.4

Can this RC be installed via the plugin manager? My plugin manager is not listing plugins at the moment for some reason - so I can't see if RC are also listed in OctoPrint's plugin manager.  Can I install the new RC version of UICustomizer via pip perhaps?

I am running `OctoPrint: 1.8.7` and the layout is a bit messed up.  Currently OctoPrint's pip lists

    UI-Customizer             0.1.9.4

crysxd, 2023-04-06:
@abulka If you are not using OctoPrint 1.9 then this is not your issue. If you want to use OctoPrint 1.9 RC, you can select it in the "Software Update" settings (select "Maintenance RC"), but this might bring you more issues than it solves

[image]

abulka, 2023-04-06:
I rebooted my pi and the plugin manager started working again. And the layout corrected itself. So all good. And I'll stay away from OctoPrint 1.9 RC till its stable etc - thanks.

WimVDK, 2023-08-27:
Still broken on Octoprint 1.9.2 and UI Customizer 0.1.9.8

LazeMSS, 2023-08-27:
@WimVDK could you explain a bit more?

LazeMSS, 2023-08-28:
@WimVDK  try disabling A dashboard for Octoprint or maybe just their webcam implantation

WimVDK, 2023-08-28:
I explicitly installed that one just to get a working webcam feed again on the home page. If this bug gets fixed I'm uninstalling it again probably. It's UI-Customizers that's not not working

jneilliii, 2023-08-28:
I still think it might be my plugin, consolidate temp control.

WimVDK, 2023-08-28:
Going to bed now. I'll see if i can confirm ly earlier statement about CTC or UIC. To be continued

WimVDK, 2023-08-29:
> I still think it might be my plugin, consolidate temp control.

I have to correct my earlier statement CTC is indeed the culprit. The moment that one was disabled and octoprint was disabled preview worked fine on all tabs. So how do we go from here?

jneilliii, 2023-08-29:
Try switching to consolidated tabs plugin instead of that one, combine temperature and control and see if it's still an issue.

WimVDK, 2023-08-29:
That one works. I'll have to tinker around with it to see which method i like best. Certainly a neat concept though! How do you remove "enabled" tabs? couldn't find that in the configuration instructions on the github/plugin settings. Seems like the "spoolmanager plugin","State","Files", "connection" and "webcam" sections can't be moved either (yet)

jneilliii, 2023-08-29:
That plugin only works on combining tabs, not sidebar items.